r-lidar/lidR #430: catalog_apply() & readTLSLAS() gives warnings (minor issue)

zoeschindler, 2021-05-19:
Hello,
I am sorry if I should have posted this in the discussion, I was not sure where to post this. Right now, I was writing some functions for `catalog_apply()` and somehow, when I use `readTLSLAS()` instead of `readLAS()`, I get warnings:
```
1: Argument 'select' is not used with a LAScluster. Use opt_select() with the LAScatalog instead.
2: Argument 'filter' is not used with a LAScluster. Use opt_filter() with the LAScatalog instead.
```
As soon as I switch back to `readLAS()` (I also tried `readALSLAS()`, it also gives warnings), everything works without a warning. I am not 100% sure, but the output looks similar to me. But I also did not give any filter or select arguments to the `readTLS()` function. I guess it's just a minor issue, it just took me some time to realize what the problem was. I just wanted to inform you guys, I am not to sure if it's relevant at all. I tried this on an <a href="https://cran.r-project.org/web/packages/lidR/vignettes/lidR-catalog-apply-examples.html">example</a> too (just switching the function how to read in the data) and it gives the same warnings. Maybe this is also intentional? I don't know.

Jean-Romain, 2021-05-19:
Please show your code so I can spot where and why you have this.

zoeschindler, 2021-05-19:
Okay, this is the code from the example page. The issue is in the `filter_noise.LAScluster()` function.

```
filter_noise.LAS = function(las, sensitivity)
{
  p95 <- grid_metrics(las, ~quantile(Z, probs = 0.95), 10)
  las <- merge_spatial(las, p95, "p95")
  las <- filter_poi(las, Z < p95*sensitivity)
  las$p95 <- NULL
  return(las)
}

filter_noise.LAScluster = function(las, sensitivity)
{
  las <- readTLSLAS(las)  # HERE, WHEN CHANGING THIS
  if (is.empty(las)) return(NULL)
  
  las <- filter_noise.LAS(las, sensitivity)
  las <- filter_poi(las, buffer == 0)
  return(las)
}

filter_noise.LAScatalog = function(las, sensitivity)
{
  opt_select(las) <-  "*"
 
  options <- list(
    need_output_file = TRUE,
    need_buffer = TRUE,
    automerge = TRUE)
  
  output  <- catalog_apply(las, filter_noise.LAScluster, sensitivity = sensitivity, .options = options)
  return(output)
}

path_points <- "H:/data/testing.las"
points_name <- "banana"

ctg <- readTLSLAScatalog(path_points)
opt_chunk_buffer(ctg) <- 0.25
opt_chunk_size(ctg) <- 20
opt_output_files(ctg) <- paste0(dirname(path_points), "/filtered/", points_name, "_{ID}")
filter_noise.LAScatalog(ctg, 1.2)
```

Jean-Romain, 2021-05-19:
Ho ok I got it. Use `readLAS()` in the `LAScluster` version. 

The idea behind the LAScatalog processing engine is to provide tools for users. And you never know if users are processing ALS TLS or whatever. So, as a developer you should never call the ALS or TLS versions of `readLAS()`. It is the responsibility of the user to choose. Inside the function you are a developer, outside your are a user.

The object `LAScluster` already carries the information needed (because you used `readTLSLAScatalog()`) and `readLAS()` will do the job transparently.

That being said you're write I should add something to trigger a warning or an error and better document this point.

Jean-Romain, 2021-05-19:
I added an error. Thanks

closed 2021-05-19 by Jean-Romain