r-lidar/lidR #431: readLAS memory issue

Saadi4469, 2021-05-19:
Hello,

While reading a large LAZ file of 1.15 GB using the readLAS function in lidR, I get the following error. Also to note, when I start RStudio, it takes 4 GB of RAM, so I don't know why I am having a memory error. Moreover, I also don't understand why lidR is bumping up memory use in Rstudio from 4 GB to 9 GB for a 1.15 GB LAZ file. RStudio is currently using 9 GB of RAM after getting this error. How can this be fixed?

```
# Code
library(lidR) 
LAS = readLAS("~")

# Available memory
memory.limit()
16283
```

Error: cannot allocate vector of size 632.4 Mb

Jean-Romain, 2021-05-19:
Please read https://cran.r-project.org/web/packages/lidR/vignettes/lidR-LAS-class.html

Your LAZ file is 1 GB so the LAS is something like 4 GB so it could easily takes up to 8 GB to read it in R. See link above.

closed 2021-05-19 by Jean-Romain

JakubCha, 2021-05-20:
LAZ is compressed LAS file. It has to be uncompressed in order to read it according to LAS specs and load into RAM memory. You can read more here:
https://laszip.org/#documentation

Jean-Romain, 2021-05-20:
Loading LAZ data does not make sense. It is not converted to LAS, it is decompress into the original data. Pretty much like you have to unzip to access the content of a zip archive.