r-lidar/lidR #433: grid_terrain() warning that there less than four coordinates causes a fatal error when running terrain()

cingularities, 2021-05-24:
Thank you for the package, very helpful! It has been running smoothly until one LAS image that resulted in a fatal error. I am having an issue with grid_terrain() and terrain(). A warning sign that states there are less than 4 coordinates at the polygon  pops up when I run grid_terrain(). Then a fatal error crashes rstudio when running terrain(). I tried running the code directly in R and I got an error for terrain() that the projection is not true. Please see code and images attached. The goal is to create CHM, DTM, slope, aspect and hillshade from [NEON LiDAR](https://www.neonscience.org/data-collection/lidar#:~:text=When%20NEON%20conducts%20an%20airborne,make%20up%20the%20top%20canopy.). Please let me know if you have issues downloading the [sample LAS scene.](https://drive.google.com/file/d/1iwrDP8l3vVuGilYhrm8g_vJUrZxyCN-3/view?usp=sharing) Any help would be appreciated!

Thank you,
Cindy

**Code**
`library(raster)
library(lidR) 
library(rgdal) 
library(sf) 
library(rgeos) 
library(sp)

#Bring point cloud into Rstudio
NEONpoints = readLAS("P:/RaBET/Raw/NEON/2019/Las/NEON_D14_SRER_DP1_510000_3518000_classified_point_cloud.las")

#Identify ground points using a cloth simulation filter algorithm
NEONpoints_classified = classify_ground(NEONpoints, algorithm = csf(sloop_smooth = TRUE, class_threshold = 0.55, cloth_resolution =  0.2, rigidness = 2))

#Separate the classified point cloud into a ground point cloud and canopy point cloud
ground_NEONpoints = filter_poi(NEONpoints_classified, Classification == 2)
canopy_NEONpoints = filter_poi(NEONpoints_classified, Classification == 1)

**#Create a grided digital terrain model
NEON_DTM = grid_terrain(ground_NEONpoints, res = 0.5, algorithm = knnidw(k = 12, p = 2))
Warning message:
In sp::Polygon(hull) : less than 4 coordinates in polygon**

**#Create slope
NEON_slope <- terrain(NEON_DTM, opt="slope",units = "radians")
(Rstudio) R session aborts 
(directly in R) Error in  .local(x, ...) : !is.na(projection(x)) is not TRUE**

#Create aspect
NEON_aspect <- terrain(NEON_DTM, opt ="aspect", units = "radians")

**Operating System and Versioning**
[Default] [64-bit] R-4.1.0
Windows 10 Pro
1.4.1106

Jean-Romain, 2021-05-25:
I cannot reproduce. `classify_ground` takes too long and take too much memory. Please produce a reproducible example with this step already performed.

That being said the dataset is already classified and the following works without issue

```r
library(lidR) 

#Bring point cloud into R
NEONpoints = readLAS("issue 433/NEON_D14_SRER_DP1_510000_3518000_classified_point_cloud.las", filter = "-drop_z_below 1000 -drop_z_above 1100")

#Create a grided digital terrain model
NEON_DTM = grid_terrain(NEONpoints, res = 0.5, algorithm = knnidw(k = 12, p = 2))

#Create slope
NEON_slope <- terrain(NEON_DTM, opt="slope",units = "radians")
```

closed 2021-05-27 by Jean-Romain

cingularities, 2021-07-07:
Thank you Jean. I am now getting an error for terrain.
NEON_slope <- terrain(NEON_DTM, opt="slope",units = "radians")
Error in .local(x, ...) : !is.na(projection(x)) is not TRUE