r-lidar/lidR #434: ROBUSTNESS: x & y should be x && y

HenrikBengtsson, 2021-06-02:
Hi, while troubleshooting some **future** devel issues regarding **lidR**, I spotted:

https://github.com/Jean-Romain/lidR/blob/82ef90be73cd3e8e86a49e8cb0d920e1fa751927/R/clusters_apply.R#L129
https://github.com/Jean-Romain/lidR/blob/82ef90be73cd3e8e86a49e8cb0d920e1fa751927/R/clusters_apply.R#L147

I think you meant to use `&&` (scalar) instead of `&` (vector).

Jean-Romain, 2021-06-02:
I can do it.  I do agress that programmatically speaking it is more correct that way but I can't see how it might be a problem. `&` and `&&` are similar enough with length 1 vector for the two expressions with `&` or `&&` to behave almost the same.

Actually in R I always use `&` unless I explicitly want the second term not being evaluated if the first is `FALSE` like in `!is.null(x) && x == 0`

HenrikBengtsson, 2021-06-02:
>  I do agree that programmatically speaking it is more correct that way but I can't see how it might be a problem. & and && are similar enough with length 1 vector for the two expressions with & or && to behave almost the same.

Sorry, I should have been more clear.  Technically, yes, both work, if and only if, LHS and RHS are of length one.  However, if one make a mistake somewhere causing one of them to have length > 1, using `&&` will eventually catch the mistake instead of propagating an unknown behavior.  That's why I proposed it.  See below for an example.  When I run revdep checks on **future**, **matrixStats**, etc. with these checks enabled, I keep spotting quite a few number of packages with such bugs.  

Current behavior in R:

```r
$ R --quiet --vanilla 
> c(TRUE, FALSE) && TRUE
[1] TRUE
> 
```

Future/eventual behavior in R:

```r
$ _R_CHECK_LENGTH_1_LOGIC2_=true R --quiet --vanilla
> c(TRUE, FALSE) && TRUE
Error in c(TRUE, FALSE) && TRUE : 
  'length(x) = 2 > 1' in coercion to 'logical(1)'
```

HenrikBengtsson, 2021-06-02:
To clarify further; I did not spot this mistake in **lidR**; I just happened to see the code while troubleshooting #435

Jean-Romain, 2021-06-02:
Understood. Thanks. I'm trying to use `&&` more consistently but `&` working every-time old habits are hard to shake off.

closed 2021-06-02 by Jean-Romain