r-lidar/lidR #436: Computing grid terrain with las catalog on new version 3.1.3 fails

ankitsagar1, 2021-06-04:
Hello JRR,

Version 3.1.3
I was computing grid_terrain with version 3.1.3, and i provided which ground classes to use.
The computation failed for las catalog, but succeed with readlas.

Version 3.0.2
The same piece of code worked for the version 3.0.2 without any problem.

Though i solved the problem, but I thought to let you know about this problem with new updated version.

Below is this code i used.       
`
ctg_full                                    <- readLAScatalog ( file_path_als )                                     
opt_filter  ( ctg_full )                <- " -drop_class 7"                                     
ctg_dtm                                   <- grid_terrain ( ctg_full,                                           
                                                                            res = 1, 
                                                                             tin() , 
                                                                             use_class = c(2L, 8L, 9L, 10L))
 `

Have a nice day.

Jean-Romain, 2021-06-04:
Thank you for reporting. You did not tell me everything. First the processing is full of red and didn't stop, meaning that you also used `opt_stop_early() <- FALSE`. Second you did not reported any error. Please redo with `opt_stop_early() <- TRUE` and report the error.

ankitsagar1, 2021-06-04:
> Thank you for reporting. You did not tell me everything. First the processing is full of red and didn't stop, meaning that you also used `opt_stop_early() <- FALSE`. Second you did not report any error. Please redo with `opt_stop_early() <- TRUE` and report the error.

Hello JRR, in the first run i put stop early = FALSE. 
In the next run put stop early = TRUE and the message was "No ground points , not possible to compute DTM" but i specifically provided the ground points and cross checked how many points are classified as ground. There were points classified as ground points in class 10. 

If you need additional information kindly ask

Jean-Romain, 2021-06-04:
Can you share one tile that fails.

ankitsagar1, 2021-06-04:
Please find the zip files as requested.

[jrr.zip](https://github.com/Jean-Romain/lidR/files/6598647/jrr.zip)

Jean-Romain, 2021-06-05:
I found the bug that was introduced in v3.1.0. The options were no longer propagated from the `LAScatalog` version of the function `grid_terrain()` to the `LAS` version of the same function. This was a seriously critical bug, thank you for your bug report

closed 2021-06-05 by Jean-Romain

ankitsagar1, 2021-06-05:
Thank you for solving the problem.