r-lidar/lidR #437: voxel_metrics() results are different with all_voxels=TRUE

zoeschindler, 2021-06-04:
Hello,

I really hope this is not some embarrassing user-side error, but I've got an issue with `voxel_metrics()`. Depending on whether I use `all_voxels=TRUE` or `all_voxels=FALSE`, it returns different results:

```
voxels <- voxel_metrics(las, length(X), res=0.1, all_voxels=FALSE)
nrow(voxels[voxels$V1 > 0 & voxels$Z > 0.65 & voxels$Z <=0.95])
> [1] 212
voxels <- voxel_metrics(las, length(X), res=0.1, all_voxels=TRUE)
nrow(voxels[voxels$V1 > 0 & voxels$Z > 0.65 & voxels$Z <=0.95])
> [1] 0
```

It also happens when I beforehand exchange the NAs with 0s:
[image]

The result using `all_voxels=FALSE` is the right one. I would attach the las-file, but I don't know how to add a 122MB las-file here.

Jean-Romain, 2021-06-04:
Well, I will need the file. I'm sure you have access to a remote storage with your company/university. Or google drive with your google account if you have one. Also if it is a 122 MB las file it will become something like 25-50 MB laz file.

zoeschindler, 2021-06-04:
[Here](https://drive.google.com/file/d/1HhW-TV6GrHPNOSQEnqI7rw0PatR76j_c/view?usp=drivesdk), I hope this works. I just put it into a zip-file.

Jean-Romain, 2021-06-04:
I found an important problem in the 3D rendering. I don't know exactly what it is but there is bug confirmed

Jean-Romain, 2021-06-05:
I still do not understand exactly why there was a problem with the join but I fixed it anyway. The output is now correct and I also enhanced the `plot` function for a better rendering. Thank you for reporting this issue.

commit 5927ee8a (2021-06-05): Fix #437
closed 2021-06-05 by Jean-Romain