r-lidar/lidR #438: Delineate Crowns

Saadi4469, 2021-06-07:
Good day Jean,

Thank you for responding to that question on gis.stackexchange. I tried sending you this via email, but your email server keeps on rejecting my email both from Outlook and Gmail. So, I would appreciate to keep privacy about the sample file that I am sharing [here](https://app.box.com/s/6yl5njreb56kpjp7ir42vmzueejp0cdv) with you. The code that I used is below:

```
# Bring in the ALS file
TP1 = readLAS("~/Sample.las")

# Then, normalize the data so that ground points are centered on 0.
TP1_DTM = grid_terrain(TP1.pmf, algorithm = knnidw(k = 8, p = 2))
TP1.normalized = normalize_height(TP1.pmf, TP1_DTM)

# Create a filter to remove points above 95th percentile of height
filter_noise = function(las, sensitivity)
{
  p99 = grid_metrics(las, ~quantile(Z, probs = 0.99), 10)
  las = merge_spatial(las, p99, "p99")
  las = filter_poi(las, Z < p99*sensitivity)
  las$p99 <- NULL
  return(las)
}

TP1_denoised = filter_noise(TP1.normalized, sensitivity = 1.2)

# Generate the Canopy Height Model (CHM)
TP1.CHM = grid_canopy(TP1_denoised, 0.5, pitfree(c(0,2,5,10,20), c(3,1.5), subcircle = 0.2))
plot_dtm3d(TP1.CHM)

# Here a single 5x5 moving window is used to apply a median filter:
win = matrix(1,5,5)
TP1.chm_s = focal(TP1.CHM, w = win, fun = median)

TP1.wtrshd = watershed(TP1.chm_s, th = 20)
TP1.ITD.wtrshd  = segment_trees(TP1_denoised, TP1.wtrshd)

# Hulls from the Watershed method
TP1_hulls_wtrshd  = delineate_crowns(TP1.ITD.wtrshd, func = .stdmetrics)

Error in sp::SpatialPolygons(hulls[["poly"]]) : is.list(Srl) is not TRUE
```

Kind Regards,

Jean-Romain, 2021-06-07:
I answered on gis.stackexchange. I keep the issue open to improve the code and handle this edge case with a more informative error.

Saadi4469, 2021-06-07:
Can I remove the file link though?

Jean-Romain, 2021-06-07:
Yes

commit 0636ab47 (2021-06-08): Fix #438
closed 2021-06-08 by Jean-Romain