r-lidar/lidR #442: Drag zoomed in LIDAR data in Plot3D window

Saadi4469, 2021-06-15:
In some softwares the user can hold down the middle mouse button to drag the visual data. How can this be done while correcting the tree tops identified by find_trees using the code below, how can I move the 3D point cloud up/down/right while zoomed in? in plot window?

`TTops.manual = find_trees(LAS_denoised, manual(LAS_TTops)) `

Jean-Romain, 2021-06-15:
Good question. I did not use this function for years but if I remember well you identify the trees with the right button, and the right button is also associated to the drag event. So, I guess drag action is overwritten by the selection action. The reason is that `manual()` was existing before the 3D rendering in lidR supports the drag action.

I guess I could put the selection with the middle button to keep the drag action.

Jean-Romain, 2021-06-15:
Which version of the package are you using ? It seems that the function no longer works...

Jean-Romain, 2021-06-15:
I fixed the error. I'd like to know which version of `lidR` and `rgl` were in use on your computer because the bug I experienced was not existing before and I did not change the function for years. So I suspect an slight modification in `rgl` itself.

Also the default button for selection is now the middle button. This automatically enabled the right button that is the default for panning in lidR.

Thank for reporting

Saadi4469, 2021-06-15:
I am using the latest version of lidR  ver. 3.1.3 and rgl ver. 0.106.8 available on CRAN. And I still have to use the right mouse button to add or flag tree tops, and the middle mouse button stretches the view.

Jean-Romain, 2021-06-15:
Yes of course. You must install the latest version from github. It is not yet on CRAN

Saadi4469, 2021-06-15:
for both lidR and rgl, or just lidR?

Jean-Romain, 2021-06-15:
Just `lidR`.This is the only source code I can update. I've no control on `rgl`.

Saadi4469, 2021-06-15:
Yes, the drag function works now thank you and cheers on fixing the error.

closed 2021-06-16 by Jean-Romain