r-lidar/lidR #443: Classifying LAS points

Saadi4469, 2021-06-16:
Hello,

How can I classify LAS points as low, medium and high vegetation in `lidR`? I was thinking of doing it after ITD and segmenting trees using `segment_trees `and then write that LAS object to file. But that didn't do it, so I am missing a step.

```
# Silva2016
LAS.ITD.Silva_p2r = segment_trees(LAS_denoised, silva2016(LAS.CHM_p2r, LAS.ttops))
# Write ITD as LAS file
writeLAS(LAS.ITD.Silva_p2r, "file_path/LAS_ITD.las", index = T)
```

Thank you,

Jean-Romain, 2021-06-16:
It depends how you define low, medium and high vegetation.

Saadi4469, 2021-06-16:
By defining a height cut off value in some function and then associating the height normalized points to these cutoff values. So basically, there would be 4 classifications (ground, low, medium and high vegetation) in total to be stored in the LAS file, and the user can see that information in the catalog file. But, I don't know how to do this in lidR.

Jean-Romain, 2021-06-16:
There is a pretty similar question[ here](https://gis.stackexchange.com/questions/382072/lidr-add-vegetation-classification-to-unique-trees).

If you want a specific answer to this question  please ask on [gis.stackexchange.com](https://gis.stackexchange.com/) with the tag `lidr`. I no longer answer to questions about lidR by email or on github because I receive too many questions from too many users. On stackexchange.com other users can also answer your questions and the answers thus provided will be public and well indexed by search engines so they can also help other users. Moreover we can provide link to already answered questions so I do not repeat myself again and again.

However you can still contact me on github to report troubleshooting (with reproducible examples) or to ask for new features or anything that is not a "how to" question.

closed 2021-06-16 by Jean-Romain