r-lidar/lidR #444: 3D Tree modeling using voxels

Saadi4469, 2021-06-16:
Hello,

Is it possible to do 3D tree modeling using voxels like shown in Figure 1 (under section 2.3) of [this ](https://onlinelibrary.wiley.com/doi/full/10.1002/ece3.4193)paper using `lidR`? Currently, I tried the code below, and the results are good, but I was still wondering if it could be taken to the next level.

```
# Voxel metrics
TLS_LAS_vox_met = voxel_metrics(TLS_LAS_denoised, ~list(N = length(Z)), 0.01) # calculate voxel metrics
plot(TLS_LAS_vox_met , color="N", colorPalette = heat.colors(50), size = 1, bg = "white")
```

Reference
Hess, C., Härdtle, W., Kunz, M., Fichtner, A., & von Oheimb, G. (2018). A high‐resolution approach for the spatiotemporal analysis of forest canopy space using terrestrial laser scanning data. Ecology and evolution, 8(13), 6800-6811.

Jean-Romain, 2021-06-16:
If you mean plotting cubes instead of points. No it is not possible with lidR because the rendering is excessively slow so I chose to display points. Now, if you put your hand into `rgl`, everything is doable

Saadi4469, 2021-06-16:
You see in that figure the points connect and are rendered to show an actual 3D tree, so you mean it's possible to pass the voxel object to rgl and plot it the  way its shown in that paper. If so then can you please share any good packages in R that have the ability to do so?

Jean-Romain, 2021-06-16:
Use must use "bare bone" `rgl` stuff.  https://stackoverflow.com/questions/13982597/plotting-voxel-image-in-r

Jean-Romain, 2021-06-16:
Finally, after many years of search, I discovered how to render many cubes more efficiently using. For example the second example I gave you was:

```r
library(rgl)
grd <- expand.grid(x=seq(0,10,2), y=seq(0,10,2), z=seq(0,10,2))
grd$dist <- sqrt(grd$x^2 + grd$y^2 + grd$z^2) # distance to coordinate 0,0,0
grd$col <- rainbow(ceiling(max(grd$dist+1)))[ceiling(grd$dist+1)]
grd$alpha <- rep(c(0.2, 1), each=nrow(grd)/2)
open3d()
for(i in seq(nrow(grd))){
  shade3d( translate3d( cube3d(col = grd$col[i]), grd$x[i], grd$y[i], grd$z[i]) , alpha=grd$alpha[i])
}
```

And can be written:

```r
library(rgl)
grd <- expand.grid(x=seq(0,10,2), y=seq(0,10,2), z=seq(0,10,2))
grd$dist <- sqrt(grd$x^2 + grd$y^2 + grd$z^2) # distance to coordinate 0,0,0
grd$col <- rainbow(ceiling(max(grd$dist+1)))[ceiling(grd$dist+1)]
grd$alpha <- rep(c(0.2, 1), each=nrow(grd)/2)
n = nrow(grd)
u = vector("list", n)
for(i in seq(nrow(grd))){
  u[[i]] = translate3d( cube3d(col = grd$col[i], alpha = grd$alpha[i]), grd$x[i], grd$y[i], grd$z[i])
}

shapelist3d(u)
```

Which is 5 to 6 times faster.

Jean-Romain, 2021-06-16:
I started doing something that looks good.

[image]

Saadi4469, 2021-06-16:
Thank you for sharing this and keep up the good work.

Cheers

Saadi4469, 2021-06-16:
I tried updating the package like you said with RStudio being run in administrative mode and empty environment. I still get the following error, how can it be fixed?

```
 cannot rename file 'C:/Users/saadb/OneDrive/Documents/R/win-library/4.0/00LOCK-lidR/00new/lidR' to 'C:/Users/saadb/OneDrive/Documents/R/win-library/4.0/lidR', reason 'Access is denied'
ERROR:    moving to final location failed
* removing 'C:/Users/saadb/OneDrive/Documents/R/win-library/4.0/lidR'
Warning messages:
1: In file.copy(savedcopy, lib, recursive = TRUE) :
  problem copying C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\00LOCK\ps\libs\x64\ps.dll to C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\ps\libs\x64\ps.dll: Permission denied
2: In file.copy(savedcopy, lib, recursive = TRUE) :
  problem copying C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\00LOCK\processx\libs\x64\processx.dll to C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\processx\libs\x64\processx.dll: Permission denied
3: In i.p(...) :
  installation of package ‘C:/Users/saadb/AppData/Local/Temp/RtmpmIXq0s/fileb8c4c172216/lidR_3.2.0.tar.gz’ had non-zero exit status 
```
Thank you

Saadi4469, 2021-06-16:
After trying to install the update with:

`Sys.setenv(R_INSTALL_STAGED = FALSE)`

I still get:

```
Warning messages:
1: In file.copy(savedcopy, lib, recursive = TRUE) :
  problem copying C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\00LOCK\ps\libs\x64\ps.dll to C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\ps\libs\x64\ps.dll: Permission denied
2: In file.copy(savedcopy, lib, recursive = TRUE) :
  problem copying C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\00LOCK\processx\libs\x64\processx.dll to C:\Users\saadb\OneDrive\Documents\R\win-library\4.0\processx\libs\x64\processx.dll: Permission denied

```

Saadi4469, 2021-06-16:
Restarting the machine fixed the permission denied error.

Saadi4469, 2021-06-16:
After updating the package to `3.2.0` and using the `find_trees ` function in manual mode, I can no longer drag the view with the left mouse button. Instead, it's again like I reported in another issue here, that I have to use the left mouse button to create a selection window, and even that window only removes a single tree top. Moreover, the middle mouse button stretches the view. And thus, the drag function is again missing. 

`remotes::install_github("Jean-Romain/lidR", ref = "devel")`

Jean-Romain, 2021-06-16:
My bad, I did not push new the code online in this branch. It should work now.

Saadi4469, 2021-06-17:
what's the new name of the function `find_trees `in this latest dev release?

Jean-Romain, 2021-06-17:
It did not change

Saadi4469, 2021-06-17:
Actually, the package was not loaded so when I tried loading it, I get the following error. Then I uninstalled the package and did a fresh installation but got this same error again.

```
Error: package or namespace load failed for ‘lidR’ in get(method, envir = home):
 lazy-load database 'C:/Users/saadb/OneDrive/Documents/R/win-library/4.0/lidR/R/lidR.rdb' is corrupt
In addition: Warning messages:
1: In .registerS3method(fin[i, 1], fin[i, 2], fin[i, 3], fin[i, 4],  :
  restarting interrupted promise evaluation
2: In get(method, envir = home) :
  restarting interrupted promise evaluation
3: In get(method, envir = home) : internal error -3 in R_decompress1
```

Saadi4469, 2021-06-17:
Restarting the R session fixes the issue.

Saadi4469, 2021-06-18:
Is it to possible to export the voxel object to ArcGIS Pro? The purpose would be to export the voxel object to ArcGIS Pro and render it there in order to overlay it on a base-map or raster as mentioned [here](https://pro.arcgis.com/en/pro-app/latest/help/mapping/layer-properties/what-is-a-voxel-layer-.html). The results from 3D voxel rendering are really cool as seen [here](https://app.box.com/s/nz9fs498533gjn82xwr5chgduoo0a6sw), but I would also like to overlay this 3D model over a base-map.

Thank you

Jean-Romain, 2021-06-18:
I've no idea if it is possible to export to ArcGIS. First we must know what kind of format are supported by ArcGIG. Butting a map in the `rgl` scene should be doable. See these resources:

- https://www.tylermw.com/adding-open-street-map-data-to-rayshader-maps-in-r/
- https://github.com/h-a-graham/rayvista

I already created publication ready figures with 3D DTM + point-cloud overlay like that:

[image]

Plotting voxels uses the same function than for point-cloud so it is not a big deal to display voxels on top a 3D map. It is only a matter of finding satellite images an applying it. Which is what the `rayvista` is doing.

Saadi4469, 2021-06-18:
Thank you for sharing this.

Jean-Romain, 2021-06-22:
I added an optimization to do not render hidden voxels. I'm closeing it is done for 3.2.0

closed 2021-06-22 by Jean-Romain

Saadi4469, 2021-06-22:
Thank you for adding this optimization. On a side note, can you please look into the comment I made [here](https://github.com/tylermorganwall/rayshader/issues/169)?

Saadi4469, 2022-09-02:
> 

Good day JRR, hope all is well, I found some more info on the `3D voxel format supported` by `ArcGIS Pro 3`. When you have have the time can you kindly look at it? Maybe then you can suggest a way to bring the cool 3D stuff that `lidR` and bring it into `ArcGIS Pro 3`. This way the product can be shared via `ESRI` based web applications with other stakeholders and the general public.

[What is a voxel layer? ](https://pro.arcgis.com/en/pro-app/latest/help/mapping/layer-properties/what-is-a-voxel-layer-.htm)

[Create a voxel layer](https://pro.arcgis.com/en/pro-app/latest/help/mapping/layer-properties/create-a-voxel-layer.htm) (this is the method that ESRI uses maybe this can give some further insight too).

Saadi4469, 2022-09-02:
@bi0m3trics thank you, yeah I know about `netCDF4` but I don't know how to convert a `3D voxel` object that `lidR` creates to a `nc` object then write that object to file.