r-lidar/lidR #446: Export classification LIDAR TO RASTER

diouck, 2021-06-21:
Hello, I will just like to export the LIDAR data according to their classification in a georeferenced raster format. . Here is the cdde that I tested to retrieve the classification:
`nonveg <- filter_poi(las, Classification != LASHIGHVEGETATION)
veg <- filter_poi(las, Classification == LASHIGHVEGETATION)

x <- plot(nonveg, color = "Classification", bg = "white", size = 3)
plot(veg, add = x)`

Jean-Romain, 2021-06-21:
Dear user of the lidR package,

This is a semi automatic message. I no longer answer to questions about lidR by email or on github because I receive too many questions from too many users. If you have questions about using the lidR package please ask on [gis.stackexchange.com](https://gis.stackexchange.com/) with the tag `lidr`. On stackexchange.com other users can also answer your questions and the answers thus provided will be public and well indexed by search engines so they can also help other users. Moreover we can provide link to already answered questions.

However you can still contact me (preferably on [github](https://github.com/Jean-Romain/lidR)) to report troubleshooting (with reproducible examples) or to ask for new features or anything that is not a "how to" question.

Regards

------

That being said, please clarrify your question. You are asking how to save a point cloud in a raster  file. So, *as is*, the anwser is *raster file are not designed to store point cloud*.

closed 2021-06-21 by Jean-Romain

diouck, 2021-06-21:
the objective is not to transform points in a raster but to create a grid with for example a resolution of 1m for all the points while retrieving the nature of their classification.

Jean-Romain, 2021-06-21:
`grid_metrics()`

diouck, 2021-06-21:
Yes that's what I ultimately use. here is the code I use to extract the different classes before assembling them. I certainly still have to dig to separate the classes in the output raster

#---Vegetation
las <- readLAS(LASfile, filter = "-keep_class  2 3 4 5 19 -drop_z_below 2 -drop_z_above 50", select = "*") 
classe3 <- grid_metrics(las,~mean(Z),0.5, filter = ~Classification == 3  )
classe4 <- grid_metrics(las, ~mean(Z), 0.5, filter = ~Classification == 4)
classe5 <- grid_metrics(las, ~mean(Z), 0.5, filter = ~Classification == 5)
classe19 <- grid_metrics(las, ~mean(Z), 0.5, filter = ~Classification == 19)
#Merge files
classe_merge <- raster::merge(classe3, classe4,classe5,classe19 )
# plot(classe_merge, col = heat.colors(50)) # some plotting
 
writeRaster(chm_s , "exportest/classe_merge.tif", format="GTiff",overwrite=TRUE)

diouck, 2021-06-21:
I don't know if it's clear enough but the idea is to transform the points into a grid with information on the classification and the height. An image that looks like this
[image]

Jean-Romain, 2021-06-21:
That's an interesting question. Please ask on gis.stackexchange as I said.