r-lidar/lidR #448: Unable to load las or laz files

denzle83, 2021-07-03:
Hi, 

I'm having trouble opening my las files. Where just last week everything was fine now nothing seems to load. I've moved files from my own dir to the lidR one, have compressed the files to LAZ and tried every conceivable method to load them but either receive:

- Error: File does not exist.
or 
- Error: Invalid header: 'description' must be a string of length < 32 in extra bytes description 

I have updated R and RStudio to the latest as well uninstalling and re-installing everything. Do you have any ideas what could be the cause?

Many thanks
Dan

Jean-Romain, 2021-07-03:
Can you share the file ?

denzle83, 2021-07-03:
Many thanks for your quick reply. I've left a OneDrive link for the las files, if you have any problems accessing it or use a different drive just let me know i'll add them to your preferred cloud.

I have found that using read.las instead of readLAS seems to allow files to be opened but then unable to run any other functions ie classify ground or normalise.

https://1drv.ms/u/s!AsVBoyyYmzOCgjvpNXNAKIMOY4xl?e=Z7uHvA

Regards
Dan

Jean-Romain, 2021-07-04:
I have a different issue.

```
 Error in nchar(extra_byte[["description"]]) : 
  invalid multibyte string, element 1 
```

Anyway it is an issue with `rlas`. I think it badly reads a 32 characters string that is not NUL-terninated . Reproducible example:

```r
h = read.lasheader("hen_faes_plot3_lidar.laz")
h
#>  [...]
#> $`Variable Length Records`$Extra_Bytes$`Extra Bytes Description`$NormalZ$description
#> [1] "Z component of a vector normal t0\032\0056\xfc\177"
```

Moved to `rlas`