r-lidar/lidR #449: z0 doesn't match z values generated by LAD and gap fraction

KatFern, 2021-07-08:
Hello!

I'm trying to calculate gap fraction and LAD from las files that have been normalized. Here is the code I'm using:
lad_ma11b<-LAD(z_ma11b, dz=1, z0=0)
gap_ma11b<-gap_fraction_profile(z_ma11b, dz=1, z0=0)

to generate values for 1 meter tall height bins starting at 0 meters. But in the outputs, z starts at .5 instead of 0:

z    |     gf
0.5 | 0.760096
1.5 | 0.678719
2.5 | 0.771611
3.5 | 0.933156
4.5 | 0.970537

Are these generated z values the midpoint of the bin? Or does the function automatically leave a buffer above the ground of half the chosen height? I'm using ground-based LiDAR so the understory is captured well by the scanner and plays an important role in my analysis, so I would like to start closer to the ground.

Any thoughts you have would be appreciated, thank you!

Jean-Romain, 2021-07-08:
It is the mid point. Anyway computing LAD with ground based lidar and this method does not make sense. The method is for ALS. You can read the referenced papers before to use a function to check out if the method used applies to your study case.

closed 2021-07-08 by Jean-Romain

KatFern, 2021-07-08:
Why do you say it doesn't make sense? I understand it's meant for a different type of lidar, but the files types are the same and I'm more interested in having good data for the understory not the canopy. Isn't it still calculating these metrics the same way?

Jean-Romain, 2021-07-08:
Yes it does calculate the same way. And this is the problem! To say it simple the method computes from top to bottom how beam are intercepted by the successive layers of canopy. By successive it means successive **for the laser beams**. In ALS top to bottom is the direction of the beam so the methods make sense (it is designed for ALS). But for TLS beams are traveling in all direction and, when pointed to the canopy, from bottom to top. So the method does not make sense for TLS. It computes dummy numbers.

Again read the original paper. It is a requirement to read and understand the methods you are using before to apply them on your data. Sometime it can make sense to apply ALS algorithm to TLS but it is relatively rare.