r-lidar/lidR #452: Feature request: function retrieve_flightlines to update LAS header

mbedward, 2021-07-30:
Thank you for the brilliant package.

The docs for the function `retrieve_flightlines` state:

> An attribute flightlineID is added in the LAS object.

This implied to me that the header would be updated as per the `add_lasattribute` function so that the flightlineID column would be written to file by a subsequent call to `writeLAS`. However, looking at the code for `retrieve_flightlines` that's not the case, Would it be possible to add this behaviour to the function or is there some reason why it is preferable to leave the header untouched?

Presently, I call `add_lasattribute(las, name = 'flightlineID', desc = "Flight line ID")` to accomplish this, but often forget.

Jean-Romain, 2021-07-30:
`retrieve_flightlines()` is an old function created before the package have half of the current capabilities. `FlightlineID` should actually be recorded in the core attribute `PointSourceID` without extra attribute. `retrieve_flightlines()` should therefore be `retrieve_pointsource()` or something like that.  This is why I never changed `retrieve_flightlines()`. My goal with this function is to create a new one that work better by updating `PointSourceID` and capable of processing a `LAScatalog`. But I did not find a way to successfully attribute consistent IDs to each flightline spread in on different files.

So I wont do what your are asking because this is not the way it should work. However considering the way it currently works your solution is the good one. Maybe `las$PointSourceID = las$flightlineID` is better and simpler.

mbedward, 2021-07-31:
That all makes sense - thank you.

closed 2021-07-31 by Jean-Romain