r-lidar/lidR #460: Error in fast_quantization

Jean-Romain, 2021-08-12:
### Discussed in https://github.com/Jean-Romain/lidR/discussions/459

<div type='discussions-op-text'>

<sup>Originally posted by **CrHvG** August 11, 2021</sup>
I get the following error :

```r
Error in fast_quantization(las@data[["Z"]], zscale, zoffset) : 
  Non quantizable value outside the range of representable values of type 'int'
```

The only other mention I see on Stack Exchange references a bug that was fixed with 3.1.2, but I am using 3.1.4 and still get an error.

Here is my code (las attached):

```r
tmp_las <- readLAS(...xxx...)
dtm = grid_terrain(tmp_las, algorithm = knnidw())
las_norm<-normalize_height(tmp_las, dtm)
```

[NIWO_2020_447000_4429000.zip](https://github.com/Jean-Romain/lidR/files/6971438/NIWO_2020_447000_4429000.zip)
</div>

Jean-Romain, 2021-08-12:
The error is not from quantization but from `grid_terrain()`. Your DTM has infinite values then the normalized point cloud also have an infinite value. I'll try to fix that. Point-cloud with complex convex shape are always problematic. I need to investigate to figure out where these infinite come from.

Jean-Romain, 2021-08-12:
After investigation there are several things to consider here.

First your DTM contained infinite values. Actually it contains `NaN`s that were wrongly converted to infinite. I fixed the conversion in 3.1.5 to preserve `NaN`s but can't yet explain why we have `NaN`s. So your DTM is now correct but we have another problem

```r
tmp_las <- readLAS(LASfile)
dtm = grid_terrain(tmp_las, algorithm = knnidw())
las_norm<-normalize_height(tmp_las, dtm)
#> Error: 1 points were not normalizable because the DTM contained NA values. Process aborted. 
```

You have one point outside the ground points and too far to be interpolated. This points is part of 3 other outlier points classified `low points` (class 7). You are better to get rid of them.

```r
tmp_las <- readLAS(LASfile, filter = "-drop_class 7")
```

cingularities, 2022-04-14:
I am having the same issue. 

Error in fast_quantization(las@data[["Z"]], zscale, zoffset) : 
  Non quantizable value outside the range of representable values of type 'int'

but I already loaded the file using filter = "-drop_class 7"

Jean-Romain, 2022-04-15:
Please open an issue with a minimal reproducible example

commit ae7d989e (2021-09-15): Convert NaN to NA  close #460