r-lidar/lidR #461: Error when using custom function in grid_metrics():   j doesn't evaluate to the same number of columns for each group

ptompalski, 2021-08-16:
Hello JR

I am developing some custom metric functions to run within grid_metrics(). All was going well until I reached a problem I cannot solve myself. 

The function should do some basic summary stats using ReturnNumber and NumberOfReturns, with some additional calculations based on the Z coordinate that happen later in the function (removed as it is irrelevant here).

When I test run the function on a single point cloud it works fine. But when run it inside grid_metrics() it fails and I get an error message that says:
Error in `[.data.table`(las@data, , if (!anyNA(.BY)) c(eval(call)), by = cells) : 
  j doesn't evaluate to the same number of columns for each group

Reproducible example:

```
library(lidR)
library(tidyverse)

echo_metrics <- function(z, ReturnNumber, NumberOfReturns) {
  
  D <- tibble(z, ReturnNumber, NumberOfReturns) 
  
  D <- D %>% 
    filter(ReturnNumber > 0, NumberOfReturns > 0) %>%    #filtering - points cannot have "0" ReturnNumber or "0" NumberOfReturns
    filter(ReturnNumber <= NumberOfReturns)              # filtering - NumberOfReturns cannot be lower than ReturnNumber
  
  
  echo_summary <- D  %>% 
    group_by(ReturnNumber, NumberOfReturns) %>%
    count() %>%
    mutate(label = 
             case_when(
               ReturnNumber == 1 ~ "First",
               ReturnNumber == NumberOfReturns ~ "Last", 
               TRUE ~ "Intermidiate"
             )
           
    )
  
  D <- D %>% 
    left_join(echo_summary, by = c("ReturnNumber", "NumberOfReturns")) %>%
    filter(!is.na(label))
  
  
  #summary stats#
  r1 <- D %>% 
    group_by(label) %>% 
    summarise(n_total = sum(n)) %>%
    mutate(perc = n_total / sum(n_total)*100) %>%
    select(-n_total) 
  r1 <- setNames(as.list(r1$perc), paste0("Perc",r1$label))
  
  return(r1)
}

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

x <- grid_metrics(las, func = ~echo_metrics(z = Z, ReturnNumber = ReturnNumber, NumberOfReturns = NumberOfReturns), res = 20)

#Error in `[.data.table`(las@data, , if (!anyNA(.BY)) c(eval(call)), by = cells) : 
#j doesn't evaluate to the same number of columns for each group

echo_metrics(z = las@data$Z, ReturnNumber = las@data$ReturnNumber, NumberOfReturns =las@data$NumberOfReturns)
#works fine

```

Any idea what am I doing wrong?

Jean-Romain, 2021-08-16:
You forgot to handle limit cases where some metrics are not defined. You missed that in some case the data frame `D` may not have 3 lines but only one or two. This is what happened in pixel 146 where `D` is made of first returns only:

```
# A tibble: 94 x 5
       z ReturnNumber NumberOfReturns     n label
   <dbl>        <int>           <int> <int> <chr>
 1     0            1               1    94 First
 2     0            1               1    94 First
 3     0            1               1    94 First
 4     0            1               1    94 First
 5     0            1               1    94 First
 6     0            1               1    94 First
 7     0            1               1    94 First
 8     0            1               1    94 First
 9     0            1               1    94 First
10     0            1               1    94 First
# … with 84 more rows
```
and `r1` is therefore:

```
# A tibble: 1 x 2
  label  perc
  <chr> <dbl>
1 First   100
```

Consequently your function returns a list of 3 elements in the 145 first pixels but at pixel 146 it returns only one element. You must ensure that all your metrics are always defined. From your code you can add the following

```r
  r1 <- setNames(as.list(r1$perc), paste0("Perc",r1$label))
  
  PercFirst = if (is.null(r1$PercFirst)) NA_real_ else r1$PercFirst
  PercIntermidiate = if (is.null(r1$PercIntermidiate)) NA_real_ else r1$PercIntermidiate 
  PercLast = if (is.null(r1$PercLast)) NA_real_ else r1$PercLast
  
  out <- list(PercFirst = PercFirst,
              PercIntermidiate = PercIntermidiate,
              PercLast = PercLast)

  return(out)
```

------

To debug this kind of problem you can `print()`

```r
  #summary stats#
  
  cat("----", i, "----\n")
  i <<- i + 1
  
  print(D)
  
  r1 <- D %>% 
    group_by(label) %>% 
    summarise(n_total = sum(n)) %>%
    mutate(perc = n_total / sum(n_total)*100) %>%
    select(-n_total) 
  
  print(r1)

# [...]

i <- 0
grid_metrics(las, func = ~echo_metrics(Z, ReturnNumber, NumberOfReturns), res = 20)
```

------

Also I think what you are trying to do is simpler that what your are doing. I might be wrong but this looks equivalent to me (but I did not study you code in details actually), but is simpler and two order of magnitude faster.

```r
echo_metrics <- function(ReturnNumber, NumberOfReturns) {
  
  filter <- ReturnNumber > 0 & NumberOfReturns > 0 & ReturnNumber <= NumberOfReturns
  ReturnNumber <- ReturnNumber[filter]
  NumberOfReturns <- NumberOfReturns[filter]
  
  n = length(ReturnNumber)
  
  first = sum(ReturnNumber == 1)/n
  intermediate = sum(ReturnNumber > 1 & ReturnNumber < NumberOfReturns)/n
  last = sum(ReturnNumber == NumberOfReturns & ReturnNumber > 1)/n
  
  out <- list(PercFirst = first, PercIntermidiate = intermediate, PercLast = last)
  
  return(out)
}
```

closed 2021-08-16 by Jean-Romain

ptompalski, 2021-08-16:
Thanks JR!
works now!!!