r-lidar/lidR #481: lidR::readLAScatalog error: Error in names(PHB) <- phblab

candelas762, 2021-09-14:
Version 3.1.4 of lidR.

I am working with new data sent to me that are from the year 2006 (idk if this can be relevant but this is the oldest data I've worked with so I am not sure if there is something that I am missing).

I am trying to read many las files as a catalog for processing it as such but when I try i get the following error:

```
ctg <- readLAScatalog(files)
#> Error: LASlib internal error. See message above.
#> Error in names(PHB) <- phblab : 
#>  'names' attribute [29] must be the same length as the vector [0]
```

I can read the files individually with readLAS() or readLASheader() but the problem comes when reading it as a LAScatalog. I have tested reading only one file as a catalog but I get the same error.

Looking at the source code for the function readLAScatalog() and running it line by line for only one file I was able to check names(PHB) and phblab and saw that they in fact differ in length but I dont know why:

```
phblab
#>  [1] "File.Signature"                    "File.Source.ID"                   
#>  [3] "Global.Encoding"                   "GUID"                             
#>  [5] "Version.Major"                     "Version.Minor"                    
#>  [7] "System.Identifier"                 "Generating.Software"              
#>  [9] "File.Creation.Day.of.Year"         "File.Creation.Year"               
#> [11] "Header.Size"                       "Offset.to.point.data"             
#> [13] "Number.of.variable.length.records" "Point.Data.Format.ID"             
#> [15] "Point.Data.Record.Length"          "Number.of.point.records"          
#> [17] "Number.of.points.by.return"        "X.scale.factor"                   
#> [19] "Y.scale.factor"                    "Z.scale.factor"                   
#> [21] "X.offset"                          "Y.offset"                         
#> [23] "Z.offset"                          "Max.X"                            
#> [25] "Min.X"                             "Max.Y"                            
#> [27] "Min.Y"                             "Max.Z"                            
#> [29] "Min.Z"                            
names(PHB)
#>  [1] "File.Signature"                    "File.Source.ID"                   
#>  [3] "GUID"                              "Version.Major"                    
#>  [5] "Version.Minor"                     "System.Identifier"                
#>  [7] "Generating.Software"               "File.Creation.Day.of.Year"        
#>  [9] "File.Creation.Year"                "Header.Size"                      
#> [11] "Offset.to.point.data"              "Number.of.variable.length.records"
#> [13] "Point.Data.Format.ID"              "Point.Data.Record.Length"         
#> [15] "Number.of.point.records"           "X.scale.factor"                   
#> [17] "Y.scale.factor"                    "Z.scale.factor"                   
#> [19] "X.offset"                          "Y.offset"                         
#> [21] "Z.offset"                          "Max.X"                            
#> [23] "Min.X"                             "Max.Y"                            
#> [25] "Min.Y"                             "Max.Z"                            
#> [27] "Min.Z"                             "Number.of.1st.return"             
#> [29] "Number.of.2nd.return"              "Number.of.3rd.return"             
#> [31] "Number.of.4th.return"              "Number.of.5th.return"   
```          
The part of the source code for the function readLAScatalog() where the error originates is this:

```
 verbose("Reading files...")
  header <- LASheader(rlas::read.lasheader(files[1]))
  crs <- crs(header)
  phblab <- make.names(names(header@PHB))
  phblab[4] <- "GUID"
  t0 <- Sys.time()
  pb <- NULL
  i <- 0
  headers <- lapply(files, function(x) {
    header <- rlas:::lasheaderreader(x)
    header <- LASheader(header)
    PHB <- header@PHB
    names(PHB) <- [phblab]
```
For some reason I can't upload files to test this error.

Jean-Romain, 2021-09-14:
What you are describing is not supposed to be. Please share reproducible data. Use a cloud storage such as Google drive, Microsoft things, WeTransfert, Dropbox or whatever. Your work or you university are also likely to offer you some solutions. Please pay attention to the amount of data you are sharing. Please pay attention to data size and do not share GB files, it takes ages to download at home and I don't have an illimited internet package. Thanks

candelas762, 2021-09-15:
Thanks for the tips. I have made a folder in Google  Drive with two las files (600 Mb) that you can  download here: https://drive.google.com/drive/folders/19YI9HuiWtFJQiM_zyRENmb73CW4I4UCn?usp=sharing

I have an update: in my personal computer I was able to completly remove R and RStudio and after reintalling them, I was able to read those files as a catalog but I can not do the same on the servers I work with (both with last version of lidR). I was considering to ask you to remove the issue but then I thought that maybe other persons can encounter the same issue of not being able to reinstall R.

Jean-Romain, 2021-09-15:
You wrote a comment on google drive

> Las files with errors in lidr 3.1.4 when read as catalog

Which version of `lidR` are you using both on your computer and server.  And which `rlas` version?

candelas762, 2021-09-15:
I am using ```lidR``` 3.1.4 and ```rlas``` 1.5.0 in all computers.

Jean-Romain, 2021-09-15:
As expected I cannot reproduce. It looks like a curse bug. Try to reinstall lidR (not R and RStudio).

closed 2021-09-17 by Jean-Romain