r-lidar/lidR #483: Error in grid_canopy when points extent (las argument) is larger than target extent (res argument)

jmmonnet, 2021-10-01:
Hi,

grid_canopy returns an error when the point extent (las argument) is larger than the target extent (res argument supplied as rasterLayer).

> Erreur dans C_rasterize(las, layout, subcircle, 1L) : 
>   C++ unexpected internal error in 'rasterize': point out of raster.

If the point cloud is cropped beforeward then the output is ok. The function grid_terrain runs without problem in any case.

Here is some code to reproduce this error:

```
# load data
data(las_chablais3, package = "lidaRtRee")
# output layout (smaller extent than point cloud)
output <- raster::raster(xmn = 974330, xmx = 974340, ymn = 6581630, ymx = 6581640, resolution = 1)
# grid_terrain is ok 
lidR::grid_terrain(las_chablais3, output, lidR::tin())
# grid_canopy is ok if point cloud is first cropped
lidR::grid_canopy(lidR::clip_roi(las_chablais3, raster::extent(output)), output, lidR::p2r())
# error in this case
lidR::grid_canopy(las_chablais3, output, lidR::p2r())
```

Jean-Romain, 2021-10-01:
To tell the truth I put this limit to be very sure to do no segfault. But I can remove the limitation that sometime annoy me too.

Jean-Romain, 2021-10-01:
There we go. I removed this limitation. Let me now if it works as expected.

By the way I did not tag you in [this discussion](https://github.com/Jean-Romain/lidR/discussions/471) because your package is on the irstea's gitlab and did not your name here. Please take note of what I wrote there. In practice it should not affect you but who knows... If you are your are using code like `las@proj4string` or `lasctg@polygons[[1]]@area` or `writeRDS(las)` please stop now and use dedicated functions

jmmonnet, 2021-10-11:
Thank you for the information regarding the upcoming changes in lidR and the fix in grid_canopy. And thank you very much for lidR. It has made my research so much easier... 

By the way I did not tag you in [ https://github.com/Jean-Romain/lidR/discussions/471 | this discussion ] because your package is on the irstea's gitlab and did not your name here. Please take note of what I wrote there. In practice it should not affect you but who knows... If you are your are using code like ***@***.*** or ***@***.******@***.*** or writeRDS(las) please stop now and use dedicated functions