r-lidar/lidR #485: Error when using readLAS function

arojas314, 2021-10-12:
Hello,

I have an issue where i am trying to read a .LAS file using readLAS() function, but I am receiving an error. I am using a windows x64 machine.

My code:
```
library(sp)
library(raster)
library(lidR)

LASfile <- ("./data/teak_data/lidarAOI.las")
las <- readLAS(LASfile, select = "xyzr", filter = "-drop_z_below 0")
```
Error message:
[image]

Thanks,
Alex

Jean-Romain, 2021-10-12:
I need the file to reproduce

Jean-Romain, 2021-10-13:
Tested on windows and everything was ok. I suggest to reinstall lidR.

arojas314, 2021-10-14:
It seems the issue was that I had conflicting packages (and/or global settings) from my old Rstudio and R install. A complete uninstall of R, Rstudio, and Rtools, as well as removing temporary files and folders in the users directory, should fix the issue.
Here is a good answer on stackexchange describing how to delete R and all associated files:
https://stackoverflow.com/questions/55204017/how-to-uninstall-r-and-rstudio-with-all-packages-settings-and-everything-else

arojas314, 2021-10-14:
@antoniobelchiorferraz 

Follow this link for a complete uninstall (on windows):
https://stackoverflow.com/questions/55204017/how-to-uninstall-r-and-rstudio-with-all-packages-settings-and-everything-else

__Quick review from that link:__
Make sure after you uninstall, to remove the directories for R and Rstudio in users Document directory:
C:\Users\%USERNAME%\Documents\R
and
C:\Users\%USERNAME%\Documents\Rstudio

Also make sure to remove the directories for R and Rstudio in this directory:
C:\Users\%USERNAME%\AppData\Local

If you don't remove those directories after uninstall, the previous dependencies are still saved. So make sure to do so.

After everything is uninstalled and you cleared those directories, then go ahead and reinstall in this order:
- R
- Rtools (https://cran.r-project.org/bin/windows/Rtools/)
- Rstudio

After you reinstalled, start by installing lidR through Rstudio, which will download all the necessary packages.

Then test by reading in a lidar file.
```
library(lidR)
lidar<-readLAS("C:/Users/..../Downloads/Kompia_Cloud_P2_buffer_10m.laz")
```

antoniobelchiorferraz, 2021-10-15:
@Jean-Romain 

and it gives the very same error on both MAC and windows after reinstalling R, Studio and Rtools  (and cleaning the related files carefully)

rror in (function (cl, name, valueClass)  : 
  assignment of an object of class “logical” is not valid for @‘proj4string’ in an object of class “LAS”; is(value, "CRS") is not TRUE

arojas314, 2021-10-15:
@antoniobelchiorferraz 

Check that GDAL or PROJ is already installed on your system, and if so, remove the paths from your environment variables.
Also, check your paths to see if you have Cygwin. If so, remove that also.

Jean-Romain, 2021-10-15:
Ok I think I know what might be the problem. The only difference that might explain your trouble is the fact that 3.2.0 no longer uses `rgdal`. Please report:

- Do you encounter trouble with all file or only some. The later case please show `readLASheader(file.las)` for both cases
- Report `sf::sf_extSoftVersion()`

```
   GEOS           GDAL         proj.4 GDAL_with_GEOS     USE_PROJ_H           PROJ 
 "3.9.0"        "3.2.1"        "7.2.1"         "true"         "true"        "7.2.1" 
```

antoniobelchiorferraz, 2021-10-15:
note that I'm reporting you results from my MAC, not Windows (I can do it later if necessary)

1- I've tried a few files and get the error message for all of them (see below readLASheader(file.las) for two examples)

2- sf package version (me 1.0-3)

3- `sf::sf_extSoftVersion()`
```
          GEOS           GDAL         proj.4 GDAL_with_GEOS     USE_PROJ_H           PROJ 
       "3.8.1"        "3.2.1"        "7.2.1"         "true"         "true"        "7.2.1" 
```

1a) 
```
readLASheader("/Users/..../plot3.laz")
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: GPS Week Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is GeoTIFF 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.1
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lasmerge (version 150701) 
File creation d/y:        154/2018
header size:              227 
Offset to point data:     399 
Num. var. length record:  2 
Point data format:        1 
Point data record length: 28 
Num. of point records:    88980 
Num. of points by return: 65329 19578 3642 402 28 
Scale factor X Y Z:       0.01 0.01 0.01 
Offset X Y Z:             202021 2027314 0 
min X Y Z:                202275.6 2028313 354.98 
max X Y Z:                202325.6 2028363 397.55 
Variable Length Records (VLR):
   Variable Length Record 1 of 2 
       Description: GeoKeyDirectoryTag 
       Tags:
          Key 33922 value 0 
          Key 1024 value 1 
          Key 1025 value 1 
          Key 2052 value 9001 
          Key 2054 value 9102 
          Key 3072 value 32620 
   Variable Length Record 2 of 2 
       Description: GeoKeyDoubleParamsTag 
       data:                 0 
Extended Variable Length Records (EVLR):  void
```
1b)
```
> readLASheader(i)
WARNING: no payload for LASF_Projection VLR with record_id 34736.
File signature:           LASF>Q 
File source ID:           20798 
Global encoding:
 - GPS Time Type: GPS Week Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is GeoTIFF 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.2
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lasmerge (version 201207) 
File creation d/y:        263/2021
header size:              227 
Offset to point data:     494 
Num. var. length record:  3 
Point data format:        3 
Point data record length: 34 
Num. of point records:    232841235 
Num. of points by return: 180175772 45814385 6364625 467522 18537 
Scale factor X Y Z:       1.37172e-06 8.64969e-07 8.7014e-08 
Offset X Y Z:             258656 392845 661.282 
min X Y Z:                258116.9 391655.4 661.282 
max X Y Z:                260139.7 394033.8 771.49 
Variable Length Records (VLR):
   Variable Length Record 1 of 3 
       Description: GeoTiff GeoKeyDirectoryTag 
       Tags:
          Key 1024 value 1 
          Key 1025 value 1 
          Key 1026 value 0 
          Key 2049 value 33 
          Key 2054 value 9102 
          Key 3072 value 32633 
          Key 3076 value 9001 
   Variable Length Record 2 of 3 
       Description: GeoTiff GeoDoubleParamsTag
```

Jean-Romain, 2021-10-15:
Ok good! Not really actually but thank you aanyway. Please show me

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
header = readLASheader(LASfile)
crs(header)
#> CRS arguments:
#> +proj=utm +zone=17 +datum=NAD83 +units=m +no_defs 
```

antoniobelchiorferraz, 2021-10-15:
that's another funny one. I've been receiving different messages lately, but here's the message since this problem started:

require(lidR)
Loading required package: lidR
Loading required package: raster
Loading required package: sp
lidR 3.2.1 using 1 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/Jean-Romain/lidR>.
Warning message:
multiple methods tables found for ‘area’

arojas314, 2021-10-15:
I receive the same message whe loading lidR version 3.2.1:
```
Warning message:
multiple methods tables found for ‘area’
```

antoniobelchiorferraz, 2021-10-15:
> epsg = epsg(header)
Error in h(simpleError(msg, call)) : 
  error in evaluating the argument 'object' in selecting a method for function 'epsg': object 'header' not found
> epsg
standardGeneric for "epsg" defined from package "lidR"

function (object, ...) 
standardGeneric("epsg")
<bytecode: 0x7faf22306f38>
<environment: 0x7faf222f6078>
Methods may be defined for arguments: object
Use  showMethods(epsg)  for currently available ones.
> lidR:::epsg2CRS(epsg)
CRS arguments: NA 
> crs = sf::st_crs(paste0("EPSG:", epsg))
Error in as.vector(x, "character") : 
  cannot coerce type 'closure' to vector of type 'character'
> crs
standardGeneric for "crs" defined from package "terra"

function (x, ...) 
standardGeneric("crs")
<bytecode: 0x7faee704d160>
<environment: 0x7faee7040600>
Methods may be defined for arguments: x
Use  showMethods(crs)  for currently available ones.
> as(crs, "CRS")
Error in as(crs, "CRS") : 
  no method or default for coercing “standardGeneric” to “CRS”

arojas314, 2021-10-15:
@Jean-Romain

```{r}
header <- readLASheader(lasfile)

epsg <- epsg(header)
epsg
#> [1] 32611

lidR:::epsg2CRS(epsg)
#> CRS arguments:
#> +proj=utm +zone=11 +datum=WGS84 +units=m +no_defs 

crs <- sf::st_crs(paste0("EPSG:", epsg))
crs
Coordinate Reference System:
  User input: EPSG:32611 
  wkt:
PROJCRS["WGS 84 / UTM zone 11N",
    BASEGEOGCRS["WGS 84",
        DATUM["World Geodetic System 1984",
            ELLIPSOID["WGS 84",6378137,298.257223563,
                LENGTHUNIT["metre",1]]],
        PRIMEM["Greenwich",0,
            ANGLEUNIT["degree",0.0174532925199433]],
        ID["EPSG",4326]],
    CONVERSION["UTM zone 11N",
        METHOD["Transverse Mercator",
            ID["EPSG",9807]],
        PARAMETER["Latitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8801]],
        PARAMETER["Longitude of natural origin",-117,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8802]],
        PARAMETER["Scale factor at natural origin",0.9996,
            SCALEUNIT["unity",1],
            ID["EPSG",8805]],
        PARAMETER["False easting",500000,
            LENGTHUNIT["metre",1],
            ID["EPSG",8806]],
        PARAMETER["False northing",0,
            LENGTHUNIT["metre",1],
            ID["EPSG",8807]]],
    CS[Cartesian,2],
        AXIS["(E)",east,
            ORDER[1],
            LENGTHUNIT["metre",1]],
        AXIS["(N)",north,
            ORDER[2],
            LENGTHUNIT["metre",1]],
    USAGE[
        SCOPE["Engineering survey, topographic mapping."],
        AREA["Between 120Â°W and 114Â°W, northern hemisphere between equator and 84Â°N, onshore and offshore. Canada - Alberta; British Columbia (BC); Northwest Territories (NWT); Nunavut. Mexico. United States (USA)."],
        BBOX[0,-120,84,-114]],
    ID["EPSG",32611]]

as(crs, "CRS")
CRS arguments:
 +proj=utm +zone=11 +datum=WGS84 +units=m +no_defs
```

antoniobelchiorferraz, 2021-10-15:
> LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
> header = readLASheader(LASfile)
> epsg = epsg(header)
> epsg
[1] 26917
> #> [1] 26917
> lidR:::epsg2CRS(epsg)
CRS arguments:
 +proj=utm +zone=17 +datum=NAD83 +units=m +no_defs 
> #> CRS arguments:
> #> +proj=utm +zone=17 +datum=NAD83 +units=m +no_defs 
> crs = sf::st_crs(paste0("EPSG:", epsg))
> crs
Coordinate Reference System:
  User input: EPSG:26917 
  wkt:
PROJCRS["NAD83 / UTM zone 17N",
    BASEGEOGCRS["NAD83",
        DATUM["North American Datum 1983",
            ELLIPSOID["GRS 1980",6378137,298.257222101,
                LENGTHUNIT["metre",1]]],
        PRIMEM["Greenwich",0,
            ANGLEUNIT["degree",0.0174532925199433]],
        ID["EPSG",4269]],
    CONVERSION["UTM zone 17N",
        METHOD["Transverse Mercator",
            ID["EPSG",9807]],
        PARAMETER["Latitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8801]],
        PARAMETER["Longitude of natural origin",-81,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8802]],
        PARAMETER["Scale factor at natural origin",0.9996,
            SCALEUNIT["unity",1],
            ID["EPSG",8805]],
        PARAMETER["False easting",500000,
            LENGTHUNIT["metre",1],
            ID["EPSG",8806]],
        PARAMETER["False northing",0,
            LENGTHUNIT["metre",1],
            ID["EPSG",8807]]],
    CS[Cartesian,2],
        AXIS["(E)",east,
            ORDER[1],
            LENGTHUNIT["metre",1]],
        AXIS["(N)",north,
            ORDER[2],
            LENGTHUNIT["metre",1]],
    USAGE[
        SCOPE["Engineering survey, topographic mapping."],
        AREA["North America - between 84°W and 78°W - onshore and offshore. Canada - Nunavut; Ontario; Quebec. United States (USA) - Florida; Georgia; Kentucky; Maryland; Michigan; New York; North Carolina; Ohio; Pennsylvania; South Carolina; Tennessee; Virginia; West Virginia."],
        BBOX[23.81,-84,84,-78]],
    ID["EPSG",26917]]
> #> PROJCRS["NAD83 / UTM zone 17N" [...]
> as(crs, "CRS")
CRS arguments:
 +proj=utm +zone=17 +datum=NAD83 +units=m +no_defs 
> #> CRS arguments:
> #> +proj=utm +zone=17 +datum=NAD83 +units=m +no_defs

ahmedshovon, 2021-10-18:
I got the same message.

Error in (function (cl, name, valueClass) : 
assignment of an object of class “logical” is not valid for @‘proj4string’ in an object of class “LAS”; is(value, "CRS") is not TRUE

Jean-Romain, 2021-10-18:
I made some tests this morning before to release 3.2.2 and version 3.2.1 from CRAN worked fine. The version 3.2.1 on CRAN for Windows R-release is a build from friday October 15th i.e. few hours only after my last test. It seems that the new binaries are not corrupted.  Please try again on windows and confirm if works. 

On Mac OS however the build is still from October 5th.

I do not know yet if I'm releasing 3.2.2 to trigger a new build. I'm waiting for your feedback

cingularities, 2022-01-24:
Hi, @Jean-Romain, when I run readLas in a loop, it causes a fatal error and aborts the program. Is there a way to avoid this?

for(fileName in NEONpoints2017){
#Bring point cloud into Rstudio
#Identify ground points using a cloth simulation filter algorithm

NEONpoints <- readLAS(fileName)

#Calculate vegetation height above the ground
NEON_DSM = normalize_height(NEONpoints, algorithm = knnidw(k = 12, p = 3), na.rm = TRUE)

#Remove points that are below 0
NEON_DSM_clean = filter_poi(NEON_DSM, Z < 45)

#plot(NEON_DSM_clean)

#Create canopy height model(raster) from the normalized canopy points
NEON_CHM = grid_canopy(NEON_DSM_clean, res = 0.5, p2r(subcircle = 0.3, na.fill = NULL))
#windows()

#plot(NEON_DSM_clean)

#Create canopy height model(raster) from the normalized canopy points

#Export the Canopy Height Model to a .tif
writeRaster(NEON_CHM,   
            filename=paste("D:/projects/RaBET/NEON_ONAQ/NEON_lidar-point-cloud-line/",substr(fileName, 43, 55), "_2017_CHM"),
            format="GTiff",
            overwrite=TRUE)
}

Jean-Romain, 2022-01-25:
Your question is not related to the content of this issue. Please open an issue with a reproducible example

mrascao92, 2022-07-28:
Hi, is this available in Portugal? I've been trying to find it at packages library but the only available version is lidR 3.2.1.

> I released 3.2.2. I hope this will fix the problem. Binaries should be available within a week