r-lidar/lidR #490: Request: Range check multiplier to get_range() as user-defined value

bi0m3trics, 2021-10-16:
Hey JR - I have a feature request (maybe the easiest one ever) but it's one that makes lidR work for other sensor types (MLS in my case), however you may not want to open this door...  

I've been using lidR to range correct the intensity of mobile lidar data, but to do so I have to go into `LAS.cpp` (lines 305-313) and remove (or creative an argument to bypass) the hard-coded range check multiplier (i.e., `3 * R_control`) to make `get_range()` work where ranges can be highly variable (as is the case with terrestrial lidar). To be completely honest, I can just keep rebuilding my own version with each release and comment out these lines of code, but it would seem to be an easy fix if the user were allowed to supply their own "multiplier" (in my case 100 instead of 3, since I'm scanning parallel to the surface and not perpendicular to it as with ALS) and then the function would work across all instances. 

Thanks! 
Andrew

Jean-Romain, 2021-10-17:
That makes sense. However I don't want to add a complex argument difficult to understand. I prefer to do something more generic and transparent. I think I could change the threshold as a function of the point source. E.g. if you read las with `readTLSLAS()` the point cloud is flagged `TLS` and I could handle your case internally by disabling this test.

bi0m3trics, 2021-10-17:
That would be perfect. In that case, I suggest dropping the check for TLS altogether...

Jean-Romain, 2022-02-08:
I'm closing this. If somebody comes with TLS data and a problem for range normalization with single position sensor I'll do something using a reproducible example. But right now it is only an hypothetical problem.

closed 2022-02-08 by Jean-Romain