r-lidar/lidR #492: bitmerge fails in some scenarios in segment_trees function

mcoghill, 2021-10-20:
I have been attempting to segment trees from the point cloud in an efficient manner. I have a very dense point cloud and have been attempting to use the `li2012()` algorithm since it operates directly on the point cloud, though this has been painfully slow in my attempts at using it. One of the options in order to speed up it's processing is to limit the maximum crown radius (the `speed_up()` option) to something below the default value of 10. I figured that I could run a different, much faster tree segmentation algorithm (e.g.: `watershed()`) and then using `delineate_crowns()` I could retrieve the maximum crown radius from there to inform the `li2012()` algorithm.

I was testing this out with one of my LAS tiles and found that the `bitmerge` option in `segment_trees()` was not working properly for me. I chose this option because in some cases, I have tiles which contain point clouds from more than 1 source, thus the `gpstime` algorithm would not be an option in my case. I also would not be able to use the `incremental` option due to having multiple tiles, so that is why I chose `bitmerge`. I believe that the reason it is failing has to do with integer overflow in R. Here is a link to download the file I have chosen, but note that this is not the only file that I have that contains this issue. The tile here has been height normalized.

https://1drv.ms/u/s!Aoq1dU0q4ri5kPV9xzpvYKOQ8W_pqA?e=q04GBr

I will now show the testing that I did to figure out the issues I had:

```
library(lidR)
LASfile <- file.path("KM1212_1483610_570280.las")
las <- readLAS(LASfile, filter = "-drop_z_below 0", select = "xyzt")

chm <- grid_canopy(las, res = 0.5, p2r(0.3))
ker <- matrix(1,3,3)
chm <- raster::focal(chm, w = ker, fun = mean, na.rm = TRUE)
algorithm <- watershed(chm)

las <- segment_trees(las, algorithm, attribute = "inc", uniqueness = "incremental")
las <- segment_trees(las, algorithm, attribute = "bit", uniqueness = "bitmerge")
las <- segment_trees(las, algorithm, attribute = "gps", uniqueness = "gpstime")

# How many trees from each uniqueness option?
sapply(c("inc", "bit", "gps"), function (x) length(unique(las@data[[x]])))

# So why is bitmerge not working? Referring to line numbers from segment_trees.R
# in below code:

# Lines 56 & 61
attribute <- "treeID"
output <- algorithm(extent(las))
las <- merge_spatial(las, output, attribute)

# Lines 87-95
xyapex <- function(x,y,z) {
  zmax <- max(z)
  j <- which(z == zmax) # allows to retrieve potentially multiple points that are all max
  if (length(j) > 1) j <- j[which.min(x[j])] # arbitrarily takes the lowest x coordinate
  return(list(x.pos.t = x[j], y.pos.t = y[j]))
}

X <- Y <- Z <- gpstime <- NULL
ids <- las@data[[attribute]]

# lines 113-135
identifyers <- las@data[, if (!is.na(.BY)) xyapex(X, Y, Z), by = ids]

xoffset <- las@header@PHB[["X offset"]]
yoffset <- las@header@PHB[["Y offset"]]
zoffset <- las@header@PHB[["Z offset"]]

xscale  <- las@header@PHB[["X scale factor"]]
yscale  <- las@header@PHB[["Y scale factor"]]
zscale  <- las@header@PHB[["Z scale factor"]]

xscaled <- as.integer((identifyers[["x.pos.t"]] - xoffset)/xscale)
yscaled <- as.integer((identifyers[["y.pos.t"]] - yoffset)/yscale)

identifyers[["IDs"]] <- lidR:::bitmerge(xscaled, yscaled)
```

Inspecting the "identifyers" data.table, we can see that there are many NaN values in the "IDs" column where there should be valid values, but the values that are in there seem to range from -1.403027e+308 to -4.317515e+307, which is why I suspect that there may be an issue here with integer overflow. I can tell that a lot of thought has been put into this option already, so I wouldn't expect that a full rewrite should be necessary, however perhaps including a check to ensure that all values in the "IDs" column are populated and there are no errant NaN values might be necessary. In cases where NaN values arise, perhaps somehow adjusting the `xscaled` and `yscaled` variables will allow the bitmerge function to produce non-NaN values. I'm not sure of a good way to go about that but maybe you would know best!

Thank you for all the hard work!

mcoghill, 2021-10-20:
I figured it out - not a problem with your coding at all, but a problem with my LAS tiles! I learned that I need to re-offset the tiles using `las_reoffset()`. I'll close this issue, and I apologize for pre-emptively opening this in the first place!

Jean-Romain, 2021-10-20:
Huum, I'll look at that. You should not have to reoffset in theory

Jean-Romain, 2021-10-21:
Fixed

```r
sapply(c("inc", "bit", "gps"), function (x) length(unique(las@data[[x]])))
#> inc  bit  gps 
#> 1075 1075 1075 
```

Jean-Romain, 2021-10-24:
Very annoying problem. The problem can be reduced to this one: why this c++ program returns nan

```cpp
#include <cstring>
#include <inttypes.h>
#include <stdint.h>
#include <stdio.h>

using namespace std;

int main()
{
  double t;
  uint64_t z = 0xfff6b1c0000e7b2b;
  memcpy(&t, &z, sizeof(t));
  printf("%" PRIu64 "\n", z);
  printf("%lf\n", t);
  return 0;
}
```

```
18444124762135231275
-nan
```