r-lidar/lidR #499: using shp_line() on a certain orientation

wiesehahn, 2021-11-07:
If I understand it correctly `shp_line()` used within `segment_shapes()` is sensitive to linear features in any orientation. Is it possible to alter the algorithm such that it detects only features which are linear in a certain orientation? 
E.g. horizontal or vertical? This would be similar `shp_hplane()`.

The use case would be e.g. to classify points as stems (vertical) or laying trees/deadwood (horizontal).

Jean-Romain, 2021-11-07:
Everything is possible but this is not currently implemented. Yet, I hardly believe it can help in segmenting stem/deadwood but who knows... lidR has been designed to allow such experiments.

I can that in the feature request list but with a low priority. On your side if you want to modify `shp_line` similalry to how `hplane` modifies `plane` feel free to make a PR.

You can prototype it easily with `point_eigenvalues` by the way

Jean-Romain, 2022-02-08:
It's finally done in v4

```r
library(lidR)
n = 400

xplane = runif(n,0,6)
yplane = runif(n,0,6)
zplane = xplane + 0.8 * yplane + runif(n, 0, 0.1)
plane = data.frame(X = xplane, Y = yplane, Z = zplane)

xhplane = runif(n,5,15)
yhplane = runif(n,0,10)
zhplane = 5 + runif(n, 0, 0.)
hplane = data.frame(X = xhplane, Y = yhplane, Z = zhplane)

tline = 1:n
xline = 0.05*tline
yline = 0.01*tline
zline = 0.02*tline + runif(n, 0, 0.1)
line = data.frame(X = xline, Y = yline, Z = zline)

thline = 1:n
xhline = 0.05*thline + runif(n, 0, 0.05)
yhline = 10 - 0.01*thline + runif(n, 0, 0.05)
zhline = 3 + runif(n, 0, 0.05)
hline = data.frame(X = xhline, Y = yhline, Z = zhline)

tvline = 1:n
xvline = 5 + runif(n, 0, 0.05)
yvline = 5 + runif(n, 0, 0.05)
zvline = 0.02*tline
vline = data.frame(X = xvline, Y = yvline, Z = zvline)

las <- rbind(plane, line, hplane, hline, vline)
las <- LAS(las)
las <- segment_shapes(las, shp_plane(k = 20), "plane")
las <- segment_shapes(las, shp_hplane(k = 20), "hplane")
las <- segment_shapes(las, shp_line(k = 20), "line")
las <- segment_shapes(las, shp_hline(k = 20), "hline")
las <- segment_shapes(las, shp_vline(k = 20), "vline")

plot(las)
plot(las, color = "plane")
plot(las, color = "hplane")
plot(las, color = "line")
plot(las, color = "hline")
plot(las, color = "vline")
```
[image]

closed 2022-02-08 by Jean-Romain

wiesehahn, 2022-02-08:
Awesome, I will give it try soon.

bi0m3trics, 2022-02-08:
I'll give it a try this afternoon. I can see tons of uses for this feature...

Jean-Romain, 2022-02-08:
I'm not has confident as you guys. It works well with real plane and real lines. But what is really planar or linear expect lakes? Certainly not trunks. Anyway if you achieve something good with that let me know :wink:

Jean-Romain, 2022-05-06:
Thank you for sharing.

jamestisdale, 2024-08-07:
> After a long time I finally managed to try it. It's not perfect yet but with some more experimenting and filtering it could work.
> 
> vertical linear features [image]
> 
> horizontal linear features [image]

I am trying to use this feature and I am curious how you got it to work so well - especially for the horizontal lines identifying downed trees.

wiesehahn, 2024-08-07:
I guess there are a lot of factors determining how well it works. point density, ground classification quality, log diameter, data acqusition quality, parameter tuning, ...
I was fiddling with this as well. I cant remember everything but try to filter as much as you can, here for exapmple we might be only interested in non-ground last return points up until 50 cm above ground if we are looking for downed trees.