r-lidar/lidR #502: Issue with Parallel processing in filter_duplicates

sarahsmithtripp, 2021-11-24:
Hi JR, 
I am filtering a lascatalog using lidR 4.0.0 and I am finding an issues  with filter_duplicates. 
I run: 
`opt_output_files(lidar_catalog) <- save_directory ### A directory to save outputs too
    opt_laz_compression(lidar_catalog) <- TRUE
    filter_duplicates(lidar_catalog)'
 and I get the error: 
`Error in serialize(data, node$con, xdr = FALSE) : 
  error writing to connection`

Which my googling suggests is an issue with the parallel processing. 

Do you get the same error on this with lidar 4.0? 

Thanks!

Jean-Romain, 2021-11-24:
Thank you for reporting. This indeed looks like a  somehow parallel processing related issue. But *parallel processing* alone bring very few information actually. What is the parallel backed strategy used ? Does it work with no parallel strategy registered ? What is your operating system ? In which folder did you try to write ? Can you share more code to see what you did ?

I made two minimal reproducible example. One is serial, one is parallel. They both work smooth on my machine. Can you confirm it does/does not work for you.

```r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile, chunk_size = 120, chunk_buffer = 0)

opt_output_files(ctg) <- "{tempdir()}/{ID}"
opt_laz_compression(lidar_catalog) <- TRUE
ctg_no_duplicate <- filter_duplicates(ctg)
```

```r
library(lidR)

library(future)
plan(multisession(workers = 4))

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile, chunk_size = 120, chunk_buffer = 0)

opt_output_files(ctg) <- "{tempdir()}/{ID}"
opt_laz_compression(ctg) <- TRUE

ctg_no_duplicate <- filter_duplicates(ctg)
```

sarahsmithtripp, 2021-11-29:
Hi JR, 

Sorry for the delay on this response. I ended up finding a workaround at the time. Originally, it looked like the default parallel processing of LidR was breaking. Now when I run the code you put above both of them work. When I briefly tested the first example last week that was breaking - but that seems to have resolved now. 

I'll let you know if I run into more parallel issues. Where you able to solve something on the backend? 

Thanks!

Jean-Romain, 2021-11-29:
Let me know. But you must know that the *default parallel processing* is sequential (i.e. not parallel). Anyway it does not mean there is no bug. Let me know if you experience any trouble.

closed 2021-11-29 by Jean-Romain