r-lidar/lidR #504: lidR version 3.2.3 rbind not working

rs806, 2021-11-28:
`rbind()` doesn't appear to be working in 3.2.3. I am getting this error: `'Error: 'rbind' is not an exported object from 'namespace:lidR''` when trying to combine 2 catalogues.

Jean-Romain, 2021-11-28:
Please provide a minimal reproducible example. The following is working to me

```r
``` r
library(lidR)
#> Le chargement a nécessité le package : raster
#> Le chargement a nécessité le package : sp
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
rbind(ctg, ctg)
#> Warning: Some tiles seem to overlap each other
#> class       : LAScatalog (v1.2 format 1)
#> extent      : 684766.4, 684993.3, 5017773, 5018007 (xmin, xmax, ymin, ymax)
#> coord. ref. : NAD83 / UTM zone 17N 
#> area        : 106266.3 m²
#> points      : 163.2thousand points
#> density     : 1.5 points/m²
#> num. files  : 2
```

<sup>Created on 2021-11-28 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

rs806, 2021-11-28:
I made a small subset of my files and it works. It has to be a problem with one of my LAS files that are contained in the catalogue I think.  I'll try to figure this out but `rbind` is working in fact.
`library(lidR)`
`ctg1a <- readLAScatalog("J:/2018/BlockE_2017/LAS_normalized/")`
`ctg2a <- readLAScatalog("J:/2018/BlockE_2018/LAS_normalized/")`
`ctgba <- rbind(ctg1a, ctg2a)`

closed 2021-11-28 by Jean-Romain