r-lidar/lidR #505: feature request: add Z & Zref to tree top output from find_trees()

rs806, 2021-11-28:
tree top point output as a 3D shapefile from `find_trees()` can be used in software like Summit Evolution Lite. Points need have the same vertical datum reference (i.e., elevation) to display properly. The canopy height value is useful too for viewing reference. The current method involves 2 steps using `find_trees()` and adding a value from an elevation raster (e.g., `Zdtm <- dtm[ttops]`). Because the normalization process assigns a Zref attribute to the normalized las object, it may be straightforward to use `find_trees()` directly on those objects (rather than las files written to disk).

Jean-Romain, 2021-11-28:
For a broader range of applications I'd say that the request consists in giving an option to return the attributes of the points that were tree apices. However I'm not sure that all algorithm are capable of retrieving the original point of the point cloud after the processing. I'll see if it is doable but I'm not sure of that.

AbdullahJirjees, 2023-03-28:
Hello @Jean-Romain 
I am facing two issue in the ITS is that I am unable to count the total number of trees, second I am unable to measure the height of a tree. any idea?
many thanks

closed 2023-08-24 by Jean-Romain