r-lidar/lidR #509: 4.0 Canopy height algorithms reordering catalog tiles

liamirwin, 2021-12-01:
Hi again JR,

I've been having some issues generating canopy height models with the lidR 4.0 beta...

It seems that when applying the pitfree() or dsmtin() algorithms through both grid_canopy and the new rasterize_canopy functions some strange rearrangement of the tiles occurs.. Interestingly it seems to work fine with p2r().

I have replicated the same issue on my colleague's machine who has also installed 4.0 and have also tried the same process on a machine running lidR 3.xx; on the non-beta version works just fine.

I've been having this issue with my catalog'd laz files but have replicated it with a tiled version of your Megaplot.laz which I will include here (390 kb): 

https://drive.google.com/file/d/1sdjbitqCC9f0mISRr4B659Ygpq7wBqVV/view?usp=sharing

Here is my reprex

``` r
# Issue: rasterize_canopy function jumbling catalog tiles for pitfree/dsmtin algos

library(lidR)

ctg <- catalog('E:/Github/lidR/rasterize_canopy/Megaplot_tiled')

chm_p2r <- rasterize_canopy(ctg, res = 1, algorithm = p2r())

plot(chm_p2r)
```

[image]

``` r
chm_pf <- rasterize_canopy(ctg, res = 1, algorithm = pitfree())

plot(chm_pf)
```

[image]

``` r
chm_pf_old <- grid_canopy(ctg, res = 1, algorithm = pitfree())

plot(chm_pf_old)
```

[image]

``` r
chm_p2r_old <- grid_canopy(ctg, res = 1, algorithm = p2r())

plot(chm_p2r_old)
```

[image]

<sup>Created on 2021-12-01 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

Thanks for your time

Liam

Jean-Romain, 2021-12-01:
Outch ! That's a big bug you found :anguished: ! I confirm the bug. I'll investigate tomorrow.

Jean-Romain, 2021-12-02:
Minimal reproducible example

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile, chunk_size = 80)
plot(ctg, chunk = T)
chm_p2r <- rasterize_canopy(ctg, res = 1, algorithm = pitfree())
plot(chm_p2r)
```

Jean-Romain, 2021-12-02:
I fixed the bug. I'm wondering how this bug may have only affected `pitfree()`. Anyway it is fixed  now for `pitfree()` and it seems the modification did not affect the rest of the code.

closed 2021-12-02 by Jean-Romain

liamirwin, 2021-12-02:
Great thanks for fixing it JR, strange issue.

By the way when I was troubleshooting it I found the same thing happened with the dsmtin() algorithm...

Jean-Romain, 2021-12-02:
Yep `dsmtin()` is only a wrapper around `pitfree()` with well chosen parameters

commit 39b31e2b (2021-12-02): Fix #509 wrong conversion from raster to data.frame