r-lidar/lidR #510: plotting segmented point clouds

krispj, 2021-12-02:
I'm trying to plot point clouds after using segment_trees() in lidR.

`plot(las, color = "treeID", colorPalette = "auto", backend = "lidRviewer" )`

Getting errora below:
 Error in if (max(id) > length(x)) stop("Index out of bound. 'id' contains wrong ids.") : 
  missing value where TRUE/FALSE needed 

Help appreciated.

Jean-Romain, 2021-12-02:
I need a reproducible example please. I cannot guess what is the problem out of nothing.

krispj, 2021-12-02:
```
library(lidR)

LASfile <- "pointcloud.laz"
las <- readLAS(LASfile)
las <- segment_trees(las, li2012(R = 3, speed_up = 5))
plot(las, color = "treeID", colorPalette = "auto", backend = "lidRviewer" )
```

Jean-Romain, 2021-12-02:
Confirmed but this an issue with package `lidRviewer` not `lidR`.

closed 2022-02-09 by Jean-Romain

krispj, 2022-02-09:
Sorry I'm still not able to run it. When I use unsegmented las file as in your code above, I get the error 
_Error: 'color' should refer to an attribute of the LAS data._

And when I used 'segmented' object as in my original code, I get the error
_Error in if (max(id) > length(x)) stop("Index out of bound. 'id' contains wrong ids.") : 
  missing value where TRUE/FALSE needed_

bi0m3trics, 2022-02-09:
Thanks. And I was just asking about the interviews out of curiosity.

Jean-Romain, 2022-02-10:
Are you sure you install the latest version from github ?

krispj, 2022-02-10:
Works now, thank you!

commit 9cf0dec5 (2022-02-09): Fix #510