r-lidar/lidR #514: cloud_metrics function returning error: Metric 'itot' is greater than 2147483647 and cannot be stored as integer.

mansi-aggarwal-2504, 2021-12-07:
Hello!
I am using the cloud_metrics function on .las files when I am getting a certain error:

[image]

Strangely, I am not getting this error on macOS and only on my windows machine, but can't be fully sure if this is an OS issue.

What is the reason for the error?

JakubCha, 2021-12-07:
I believe it is the same issue as in #463 but JR added specific error that has printed in your case.
You should scale down intensities in your point cloud as JR pointed out here https://github.com/r-lidar/lidR/issues/463#issuecomment-905422994 and scale them up after computing them (if it makes sense in your case).

Jean-Romain, 2021-12-07:
This is indeed the same problem as in #463. I changed `itot` so it is now always `double` so it should no longer be a problem. You can also scale the intensity as mentioned by @JakubCha

closed 2021-12-07 by Jean-Romain

mansi-aggarwal-2504, 2021-12-10:
@JakubCha @Jean-Romain thank you for your help. I'll try this.
However, I don't understand why I am getting this issue on one system and not on another. Where does the OS dependency factor in?

Jean-Romain, 2021-12-10:
That's very weird indeed. No it should not be OS dependent.. But it is impossible to answer without a reproducible example