r-lidar/lidR #521: Installing on Ubuntu: ERROR: dependencies ‘sf’, ‘RcppArmadillo’ are not available for package ‘lidR’ * removing ‘/home/user/R/x86_64-pc-linux-gnu-library/4.1/lidR’

mavavilj, 2021-12-28:
Installing on Ubuntu 20.04.3 LTS the latest from Github using:

```
# Ubuntu
sudo add-apt-repository ppa:ubuntugis/ubuntugis-unstable
sudo apt-get update
sudo apt-get install libgdal-dev libgeos++-dev libudunits2-dev libproj-dev libx11-dev libgl1-mesa-dev libglu1-mesa-dev libfreetype6-dev libxt-dev libfftw3-dev
```

and then

```
devtools::install_github("Jean-Romain/rlas")
```

goes OK, but

```
devtools::install_github("Jean-Romain/lidR")
```

I end up to:

```
Installing package into ‘/home/user/R/x86_64-pc-linux-gnu-library/4.1’
(as ‘lib’ is unspecified)
ERROR: dependencies ‘sf’, ‘RcppArmadillo’ are not available for package ‘lidR’
* removing ‘/home/user/R/x86_64-pc-linux-gnu-library/4.1/lidR’
Warning messages:
1: In i.p(...) :
  installation of package ‘classInt’ had non-zero exit status
2: In i.p(...) :
  installation of package ‘RcppArmadillo’ had non-zero exit status
3: In i.p(...) : installation of package ‘sf’ had non-zero exit status
4: In i.p(...) :
  installation of package ‘/tmp/Rtmp8WZQid/file2199f292ec5f4/lidR_3.2.4.tar.gz’ had non-zero exit status
> 
```

Jean-Romain, 2021-12-28:
The CRAN provides binary versions of the packages for Windows because it is so difficult to install packages on windows that someone has to do the job for Windows users. This is why it is easy on Windows. On linux it is much easier so nobody is doing the job for you. As a consequence it is more difficult...

That being said your problem is related to `sf` and `RcppArmadillo`. I'm not the maintainer of these packages and the message you copy pasted only tells that installation of `classInt` `RcppArmadillo` and `sf` failed. Try to install `classInt` alone and look at the error. Then `RcppArmadillo`. When both will be ok `sf` will be ok as well I guess

mavavilj, 2021-12-28:
I did some manual messing around and got RccpArmadillo installed by following instructions here:

https://tommasorigon.github.io/CompStat/exe/un_A2.html

I still didn't install sf, but when I typed

`install.packages("lidR")

lidR is not working properly.

But it might be good idea to add more complete instructions for installing on Ubuntu 20.04.
`

Jean-Romain, 2021-12-28:
It is not my role to explain how to install other R packages on Linux or to tech how R works on Linux. I'm only maintaining lidar related packages and there is already tons of resources on internet :wink:  I'm glad you solved your problem.

closed 2021-12-28 by Jean-Romain