r-lidar/lidR #522: gap_fraction_profile function maybe have some problems?

zzk112233, 2022-01-11:
Hi author, lidR is a really nice R package to use. But recently I seem to have noticed some problems with the **gap fraction profile**  function. Hope to get your answer.
According to your reference  _Bouvier, M., Durrieu, S., Fournier, R. a, & Renaud, J. (2015). Generalizing predictive models of forest inventory attributes using an area-based approach with airborne las data. Remote Sensing of Environment, 156, 322-334. http://doi.org/10.1016/j.rse.2014.10.004_.
p(z) is gap fraction, P = N[0: z]/(N[total]-N [0: z dz])；with N[0;z] being the number of returns below z, Ntotal is the total number of returns, N[0;z + dz] is the number of returns below z + d z. But in your function, P = (N [0: z]/N [total])/(N [0: z+dz]/N [tatol]). Maybe P = (N [0: z]/N [total])/1-(N [0: z+dz]/N [tatol]) is the right thing?
#gap_fraction_profile function
function (z, dz = 1, z0 = 2) 
{
    bk <- seq(floor((min(z) - z0)/dz) * dz + z0, ceiling((max(z) - 
        z0)/dz) * dz + z0, dz)
    if (length(bk) <= 1) 
        return(data.frame(z = numeric(0), gf = numeric(0)))
    histogram <- graphics::hist(z, breaks = bk, plot = F)
    h <- histogram$mids
    p <- histogram$counts/sum(histogram$counts)
    p <- c(p, 0)
    cs <- cumsum(p)
    **_i <- data.table::shift(cs)/cs_**
    i[is.na(i)] = 0
    i[is.nan(i)] = NA
    z = h
    i = i[-length(i)]
    return(data.frame(z = z[z > z0], gf = i[z > z0]))
}
maybe **_i<- data.table::shift(cs)/1-cs_** is correct?
Looking forward to your answers. I'm new to Lidar, so I'm sorry if I offended.

Jean-Romain, 2022-01-11:
No offense here but the formatting of your question make it difficult to read so I'm not sure of the problem you are trying to highlight. The point is that the equation 5 in Bouvier et al. is just wrong. It does not match with the text. Read carefully the text: 

> This required assessing the number of laser beams that actually reached the layer z+ dz and those that went through the layer [z; z+dz]. Eq. (4) was thus modified, and Pi and LAIi for the layer i between z and z+dz were

think of the physical meaning and you will see it does not match equation 5. I reported the problem to the authors several years ago. A correct version of the equation with more details on the physical meaning has been published in https://www.sciencedirect.com/science/article/abs/pii/S0034425717305849 and is:

![](https://quicklatex.com/cache3/23/ql_3c5cfc13a95e978b48318566a3394b23_l3.png)

zzk112233, 2022-01-11:
Thank you for your reply. It really answers my question. I have another question here. Can I use this LAD function to calculate the lidar data for hand-held mobile laser scanning, which emits laser beam in a direction different from that of ALS?

Jean-Romain, 2022-01-11:
No, Bouvier's equations make sense only for ALS.

closed 2022-01-11 by Jean-Romain