r-lidar/lidR #531: moving window with flexible function

spono, 2022-02-11:
Ciao JR,
I was thinking that a possible idea to enhance `find_trees` would be giving it the possibility to accept more functions (or a set of parameters) according to a second variable provided along with the height from CHM.

Basically, if I provide a stack with "height" + "forest_type" (conifer/broadleaf, high forest/coppice, etc.), the moving window applies a function according to a rule based on the ancillary layer. This would allow to be a bit more specific already during the process instead of -e.g.- running twice the algorithm for the different areas.

I guess it could be something like this:

```
params <- data.frame(a = c(0.05, 0.07), b = c(1.5, 2))

f <- function(x,y,z) { x * y + z }

find_trees(chm, layer2, lmf(f(a,b), params, hmin=2, shape = "circular"))
```

but I don't know how complex this would be to implement on the actual tools.

Jean-Romain, 2022-02-11:
Could you be more specific? It sounds like a interesting request but I did not understand the details. Why `f` has 3 parameters but only 2 are used in `f(a,b)`? How values `a` and `b` are selected? How does the "forest type" is evaluated at run time? Using `merge_spatial` and a shapefile (`layer2`?)? In this case I could design something more generic like:

```r
f <- function(x,y,z) { x * y + z }
find_trees(chm, lmf(f(Z, Intensity, ScanAngleRank)))
```

Obviously `Z * Intensity + ScanAngle` cannot give a good windows but something like that becomes possible

```r
f <- function(x,y) { ifelse(y == 1, 0.05*x + 1.5, 0.07*x + 2) }
las <- merge_spatial(las, layer, "foresttype")
find_trees(chm, lmf(f(Z, foresttype)))
```

Jean-Romain, 2022-02-11:
> it was just a draft, I didn't really mean to write a working example

Certainly! I understand  that. But a syntactically valid example helps for the understanding :wink: 

Below a quick and dirty proof and concept tweaking the code of `lmf`. It does what I described above. It works with `v4`  because I used the function `plugin_itd()` which does not exist in previous version

```r
lmf_extended = function(ws, hmin = 2, shape = c("circular", "square"), ws_args = NULL)
{
  shape <- match.arg(shape)
  circ  <- shape == "circular"
  ws    <- lazyeval::uq(ws)
  hmin  <- lazyeval::uq(hmin)
  
  f = function(las)
  {
    if (is.function(ws))
    {
      if (!is.null(ws_args))
      {
        args <- lapply(ws_args, function(x) if (x %in% names(las)) las@data[[x]] else x)
        ws <- do.call(ws, args)
        b <- las$Z < hmin
        ws[b] <- min(ws)
      }
      else
      {
        ws    <- ws(las@data$Z)
        b <- las$Z < hmin
        ws[b] <- ws(hmin)
      }
      
      n <- npoints(las)
      if (!is.numeric(ws)) stop("The function 'ws' did not return a correct output. ", call. = FALSE)
      if (any(ws <= 0))    stop("The function 'ws' returned negative or null values.", call. = FALSE)
      if (anyNA(ws))       stop("The function 'ws' returned NA values.",               call. = FALSE)
      if (length(ws) != n) stop("The function 'ws' did not return a correct output.",  call. = FALSE)
      
    }
    else if (!is.numeric(ws))
    {
      stop("'ws' must be a number or a function", call. = FALSE)
    }
    
    lidR:::force_autoindex(las) <- lidR:::LIDRGRIDPARTITION
    return(lidR:::C_lmf(las, ws, hmin, circ, lidR:::getThread()))
  }
  
  f <- plugin_itd(f)
  return(f)
}

LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile)
ws <- function(x,y) { 0.1*x + 0.01*y }

algo <- lmf_extended(ws,  ws_args = list(x = "Z", y = "Intensity"))
u = locate_trees(las, algo)               
plot(sf::st_geometry(u))

algo <- lmf_extended(ws,  ws_args = list(x = "Z", y = 100))
u = locate_trees(las, algo)               
plot(sf::st_geometry(u))
```

spono, 2022-02-11:
whoa, amazing!
unfortunately I won't be able to test it soon but I crave for it. I'll give you some feedback asap.
In the meantime, thank you very much, as usual

Jean-Romain, 2022-02-16:
Added in v4.0.0 with minimal modification + backward compatibility

closed 2022-02-16 by Jean-Romain