r-lidar/lidR #534: lasclipRectangle() doesn't get the correct xleft value I pass to it

Dafeng-Zhang, 2022-02-16:
Hi Sir,

Here is my code to clip a large TLS file into a 100m by 100m tile. I checked the TLS data I used and I am sure it has the correct xleft I want to pass to the clip function. But the output is incorrect: the minimum x coordinate of the new tile is -743576.3, rather than -743582.3. Do you know why this happens? Thank you so much! 

library(lidR)
library(rlas)
T<-readLAS("/users/dzhang53/scratch/factors/leafonTLS_classified_fromMartin/ZF_34_2018_on_vege.laz")
info_on<-read.lasheader("/users/dzhang53/scratch/factors/leafonTLS_classified_fromMartin/ZF_34_2018_on_vege.laz") #plot tls extent 
#print(info_on$'Min X')  
#-743582.3
tile1 <- lasclipRectangle(T, xleft=info_on$'Min X', ybottom=info_on$'Min Y', xright=-info_on$'Min X'+100, ytop=info_on$'Min Y'+100)
writeLAS(file=paste("/users/dzhang53/scratch/factors/","TLS_Tile_1_new.las",sep=""), las=tile1)

D

Jean-Romain, 2022-02-16:
Impossible to say without a reproducible example. First of all `lasclipRectangle()` is deprecated since 1 year and a half. And it throw an error since 6 month at least. Since how long you did not update your packages? Anyway this is not related to your problem. With a recent version of lidR what are the output of

```r
print(TLS)
print(tile1)
range(TLS$X)
las_check(TLS)
las_check(tile1)
```

Dafeng-Zhang, 2022-02-17:
Thank you for you reply. I printed TLS but it failed due to memory issue. I'll try the rest lines. Since lasclipRectangle() is deprecated, is there a new function that can do the same/similar thing? Thank you!
[image]

Jean-Romain, 2022-02-17:
It has been renamed `clip_rectangle`. If you don't even have a message it means you are running lidR version <= 2.2.5

If your file does not fit in memory use a `LAScatalog` to perform the extraction. I advise you to read the [lidR's book](https://r-lidar.github.io/lidRbook/) and [lidR's vignettes](https://cran.r-project.org/web/packages/lidR/index.html)

Your session crashed, this is a serious bug that must be reported with a reproducible example (and a recent version of the package)

Please do not include screenshot when a copy pasting is simpler and more readable.

Dafeng-Zhang, 2022-02-18:
Hi Sir,

It turned out the TLS data I used is a little weird. The point with the Min X is outside my region of interest. This is why I cannot get the xleft I wanted. I moved my ROI rightwards and it worked correctly. The memory issue is due to the memory limit for each user in a computer cluster I used.  So nothing wrong with the function. Appreciate all your suggestions! Thanks a lot!