r-lidar/lidR #539: How to install on MacOS?

kevinpauli, 2022-02-25:
Hello, I am unable to get this installed on MacOS Monterey 12.2.1 with conda `4.11.0`.

This is my environment.yml
```
name: r-lidr-test
channels:
  - conda-forge
  - nodefaults
dependencies:
  - r-base
  - r-lidr
```

`conda env create` results in:
```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package r-base conflicts for:
r-lidr -> r-base[version='3.4.1.*|>=3.4.1,<3.4.2.0a0']
r-lidr -> r-data.table -> r-base[version='3.3.2.*|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0|>=4.0,<4.1.0a0|>=4.1,<4.2.0a0|>=3.5.1,<3.5.2.0a0']
r-base
```

If I try with mamba `0.15.3`:  `mamba env create` I get this:
```Looking for: ['r-base', 'r-lidr']

Encountered problems while solving:
  - nothing provides compiler-rt 4.0.1 hcfea43d_1 needed by clang-4.0.1-1
```

Any help is much appreciated!

Jean-Romain, 2022-02-25:
I don't know what is `conda`, I don't know what is `mamba` and I don't have MacOS. Did you considered installing R and R packages the regular way? 

As far as I can understand you have troubles with `conda` not with `lidR`. You should ask how to work with `conda` to `conda` people. I'm sure you will find knowledgeable people on stackoverflow. On my side I can't help you sorry.

closed 2022-02-25 by Jean-Romain

Jean-Romain, 2022-02-25:
I didn't know the existence of this repo. I'm maintaining R packages on CRAN. Other places are mirror or package automatically generated by third parties. I not related in any way to this conda stuff. However I can see that Linux and Window have lidR v4.0.0 released few days ago while MacOS seems to have version 1.5.0 that is 4 years old!