r-lidar/lidR #548: Problems with normalize_intensity and track_sensor

cflynn8, 2022-03-11:
I am trying to use lidR for the intensity normalization function it offers and I am consistently running into trouble. I have several different ALS datasets that I have tried with and am hoping to get some direction to figure out what is going wrong. I can provide sample .las files if needed. 

Here is the code in RStudio:

library(lidR)
library(future)
## Normalize intensity of a single las file
las1 <- readLAS("S:\\IRAD\\LidarProcessing\\IntensityNormalizationTesting\\SaltonSeaTestData\\LAS_Copy\\11SPS180600.las", filter = "-drop_withheld")
plot(las1, color = "Intensity")
las1 <- filter_duplicates(las1)
sensor <- track_sensor(las1, Roussel2020(pmin = 50))
las_norm <- normalize_intensity(las1, range_correction(sensor, Rs = 2000))
plot(las_norm, color = "Intensity")

And here is the error I am currently getting (at the line: las_norm <- normalize_intensity(las1, range_correction(sensor, Rs = 2000))):
3 coordinates detected in the sensor positions, parameter 'elevation' is not considered.
Error: 'gpstime range from the sensor does not contain gpstime range from the point-cloud

Any help is greatly appreciated!

Jean-Romain, 2022-03-12:
The code looks legit. My guess is that few first or last sensors locations where missed for unknown reason. I need a file to reproduce and figure out if the internal code requires improvements of if it comes from your file

cflynn8, 2022-03-14:
Thank you for the quick response. Here is the same las file I tried to test with in that code example. [https://gmuedu-my.sharepoint.com/:f:/g/personal/cflynn8_gmu_edu/Eup5N1vIuXRDmMCRaRETr7IBnnfNi4AicR6aemTaKP1G7w?e=EbYcFC]

Jean-Romain, 2022-03-14:
I see

```r
range(sensor$gpstime)
[1] 320787224 320962446
range(las1$gpstime)
[1] 320787222 320962446
```

As expected the very first measurement is missing probably because there was not enough data to compute anything at the edge of the tile. This should not prevent to use the function `normalize_intensity` so I will add some tolerance tomorrow.

Jean-Romain, 2022-03-14:
I fixed your issue.  But here some additional comment

1. Your dataset is flat no need to apply a range correction there is nothing to correct
2. Without a buffer and extra tiles on the edges the accuracy of the sensor detection is not super accurate. Especially there are 2 outliers at 400 and 700 m height.
3. Your intensity range between 0 and 65535 i.e. the maximum storable value of intensity. When applying the range correction most of the points see their intensity lowered (because `Rs = 2000` i.e. higher than the actual sensor) but some see their intensity increased. But it is not possible because the new values are above the maximum. Consequently you have many warnings.

closed 2022-03-14 by Jean-Romain

Jean-Romain, 2022-03-15:
> reduce the intensity of points collected over water at nadir that result in very high intensity return values.

You apply a range corrections. Not  a water correction

> I do not see any option to update the package in RStudio either

```r
remotes::install_github("r-lidar/lidR")
```

cflynn8, 2022-03-21:
Thank you! That install worked for me and the tool ran fine on that tile. I tried to build slightly from there and run it with an LAS catalog of just 9 tiles. It started to process the area and one of the tiles completed. However, it failed on the tile that I had been testing individually (and supplied here). Is there any reason that the tool would fail when processing an LAS catalog?

Also, is there a water correction? I had not seen one.

Thanks!

Jean-Romain, 2022-03-21:
I cannot know without a reproducible error. Please open a new issue explaining what is going wrong.

candelas762, 2022-07-15:
> 1. Your dataset is flat no need to apply a range correction there is nothing to correct

Hi, I was wondering what you meant by "flat". Is it that is normalized? I checked the file and indeed it seems to be normalized (although heights are below 0). In this case I have the following question, is not correct to normalize intensity if the LAS file is already normalized?