r-lidar/lidR #554: TreeIDs of identified trees and crowns

nbrodic88, 2022-03-23:
HI!
Would like to know if function segment_trees with Silva gives other treeID-s than the ones I got with function find_trees?
Regards,
Nenad

Jean-Romain, 2022-03-23:
Did you try ?

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyz")
col <- pastel.colors(200)

# Using raster because focal does not exist in stars
chm <- rasterize_canopy(las, res = 0.5, p2r(0.3), pkg = "raster")
ker <- matrix(1,3,3)
chm <- raster::focal(chm, w = ker, fun = mean, na.rm = TRUE)

ttops <- locate_trees(chm, lmf(4, 2))
crown <- silva2016(chm, ttops)()

plot(crown == 150)
plot(sf::st_geometry(ttops[ttops$treeID == 150,]), add = T, pch = 19)
```

The seed IDs are preserved. If called in `segment_trees` the IDs are preserved too.

```r
las = segment_trees(las, silva2016(chm, ttops))

plot(filter_poi(las, treeID == 150))   |>
  add_treetops3d(ttops[ttops$treeID == 150,])
```

nbrodic88, 2022-03-23:
You're right, segmented tree from LiDAR point cloud and detected treetop has the same treeID.
[image]

But when I calculate crown metrics it numerates tree crowns with different treeid and also it looks like that the metrics are not related to the generated geometry:
crowns <- crown_metrics(las, func = .stdmetrics, attribute = "treeID", geom = "concave")
[image]

and is there a difference between .stdmetrics and .stdtreemetrics because with .stdtreemetrics I don't get the all the attributes for each segment

Jean-Romain, 2022-03-23:
`crow_metrics` is supposed to respects the IDs. If you think something is wrong please report a minimal reproducible example. I cannot guess was is supposedly wrong with your image that looks legit.

Jean-Romain, 2022-03-23:
I can't see any obvious problem here. I may have missed it but if you think something is wrong please **open a new issue** with a minimal reproducible example and **explain clearly what does not correspond to your expectation** using code or image. Thanks

hansoleorka, 2022-04-10:
Hi

I noticed a similar issue using the watershed algorithm. I modified the above example. It seems like there is an issue with the treeID when using crown_metrics. See the resulting plot of the code below. 

Regards
Hans Ole  

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyz")
col <- pastel.colors(200)

chm <- rasterize_canopy(las, res = 0.5, p2r(0.3), pkg = "raster")
ker <- matrix(1,3,3)
chm <- raster::focal(chm, w = ker, fun = mean, na.rm = TRUE)

ttops <- find_trees(chm, lmf(4, 2))
las = segment_trees(las, silva2016(chm, ttops))
crowns <- crown_metrics(las, func = .stdmetrics_z, attribute = "treeID", geom = "concave")

tree <- filter_poi(las, treeID == 80)
top <- ttops[ttops$treeID == 80,]
crown <- crowns[crowns$treeID == 80,]

plot(chm)
plot(sf::st_geometry(crown),add=T)
plot(top,pch=3,add=T)
points(tree$X,tree$Y,pch=".")
```

Jean-Romain, 2022-04-10:
Thank you for the reproducible example I conform there is something weird. I'll investigate tomorrow.

Jean-Romain, 2022-04-11:
It is fixed now

closed 2022-04-11 by Jean-Romain

Jean-Romain, 2022-04-12:
from github

nbrodic88, 2022-04-12:
Now it works great, even when you import treetops from external file

spono, 2023-11-30:
Ciao JR,
it seems that, in v 4.0.4, using the standalone algorithm (e.g.  `dalponte2016(chm, ttops)()` ) the `treeID` of crowns is renamed differently (in my case `Z`), while using `segment_trees` and then `crown_metrics` leads to a correct output.

Updating a bit the code from Hans Ole, you get:

```
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyz")
col <- pastel.colors(200)

chm <- rasterize_canopy(las, res = 0.5, p2r(0.3), pkg = "raster")
ker <- matrix(1,3,3)
chm <- raster::focal(chm, w = ker, fun = mean, na.rm = TRUE)

ttops <- locate_trees(chm, lmf(4, 2))

crowns <- dalponte2016(chm, ttops)()
names(crowns)
> "layer"
```

Nevertheless, on a whole segmented LAScatalog, tree IDs and crown IDs seem not to correspond to each other...but I'm digging in more and, in case, I'll open a separate issue.

commit 984e1ff7 (2022-04-11): Fix #554