r-lidar/lidR #556: Error in normalize_intensity: only point geometries are supported

cflynn8, 2022-03-29:
I am trying to use normalize_intensity on a small set of las files. I am linking to the data [here](https://gmuedu-my.sharepoint.com/:f:/g/personal/cflynn8_gmu_edu/EvobcXVbUgREjWlrKR1aK-UBRwwH9d2puHa4rfurw63w0Q?e=ckpjZJ). Here is my r code:
```
# Normalize intensity of a folder of las files
library(lidR)
library(future)
plan(multisession)

# Read in the folder of las files as and LASCatalog
marked_cat <- readLAScatalog("S:\\IRAD\\LidarProcessing\\IntensityNormalizationTesting\\SaltonSeaTestData\\LAS_Copy\\LAS_SmallTiles\\marked",
                             filter = "-drop_withheld -drop_class 7 18")

las_check(marked_cat)

# Visualize the las tiles of the files being processed
plot(marked_cat)

# Estimate the positioning of the lidar sensor. Uses first and last returns with gpstime.
sensor <- track_sensor(marked_cat, Roussel2020(pmin = 50))

# Set a location to store the output files with normalized intensity values
opt_output_files(marked_cat) <- "S:\\IRAD\\LidarProcessing\\IntensityNormalizationTesting\\SaltonSeaTestData\\LAS_Copy\\LAS_SmallTiles\\marked\\norm"
  
# Normalize the intensity using the sensor positioning from above.
las_norm <- normalize_intensity(marked_cat, range_correction(sensor, Rs = 2000))
```
From the last line of code, I get the following error:
Error: Only point geometry types are supported

The files I am reading in are las files (v 1.4).

Thank you!

Jean-Romain, 2022-03-29:
Please show the output of `marked_cat`, `las_check()` and `sensor`

cflynn8, 2022-03-29:
`marked_cat` does not output anything in the console but reads in to the environment as a "Formal class LAScatalog".

`las_check(marked_cat)` output:
```
 Checking headers consistency
  - Checking file version consistency... ✓
  - Checking scale consistency... ✓
  - Checking offset consistency... ✓
  - Checking point type consistency... ✓
  - Checking VLR consistency... ✓
  - Checking CRS consistency... ✓
 Checking the headers
  - Checking scale factor validity... ✓
  - Checking Point Data Format ID validity... ✓
 Checking preprocessing already done 
  - Checking negative outliers...
    ⚠ 4 file(s) with points below 0
  - Checking normalization... no
 Checking the geometry
  - Checking overlapping tiles... ✓
  - Checking point indexation... no
```
sensor output:
```
Warning messages:
1: In min(bb[, 1L], na.rm = TRUE) :
  no non-missing arguments to min; returning Inf
2: In min(bb[, 2L], na.rm = TRUE) :
  no non-missing arguments to min; returning Inf
3: In max(bb[, 3L], na.rm = TRUE) :
  no non-missing arguments to max; returning -Inf
4: In max(bb[, 4L], na.rm = TRUE) :
  no non-missing arguments to max; returning -Inf
```
[image]

Jean-Romain, 2022-03-30:
I can probably enhance lidR by triggering an informative error but your problem is pretty simple: you have 6 points from 2 pulses with more than one returns over the entire coverage. And two files have literally 0 multiples returns.

```r
las = readLAS(marked_cat, filter = "-drop_withheld -drop_class 7 18")
filter_poi(las, ReturnNumber > 1)
#> class        : LAS (v1.4 format 6)
#> memory       : 4.2 Kb 
#> extent       : 620100.2, 620400, 3661536, 3661642 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NA 
#> area         : 4873.4 units²
#> points       : 6  points
#> density      : 0 points/units²
```

Jean-Romain, 2022-03-30:
Your example no longer fails and returns an empty `sf data.frame` and triggers a warning.

commit 67d6092b (2022-03-30): Fix #556
closed 2022-03-30 by Jean-Romain