r-lidar/lidR #557: Plot Fatal Error

cingularities, 2022-03-29:
When I Plot las points I get a fatal error and Rstudio shuts down. This problem just started a few days ago. 
Thank you

Jean-Romain, 2022-03-30:
This is not very informative and helpful. Which OS? Which version of `lidR`? Which version of `rgl`? Did you update some packages few days ago? What happended few days ago? Minimal reproducible example? Does `rgl` rendering works outside of `lidR`? Which version of R? Does it crash with every point cloud or only some?

cingularities, 2022-03-30:
Hello Jean-Romain,
Thank you for the quick response. I did not update any packages before issues started. I had IT update Rstudio version and I updated my packages yesterday but fatal error persists.

I am running on Windows 14393.5006 64 bit Intel(R). My Rstudio: Version 1.4.1717 with R version 4.1.0 (2021-05-18) and lidR 3.2.4. I did not install rgl separately. I was running as a simple plot(points_georeferenced) and this fatal error occurs with any point cloud. I installed rgl today and tried running rgl::plot3d(points_georeferenced@data) and it worked visualizing a simple 3D plot. I ran lidR::plot(points_georeferenced@data) and it did not give me a fatal l error but kept trying to plot and I waited for 15-30 minutes with no success. I stopped the run. Please let me know if you need more details. 

Thank you very much,
Cindy

Jean-Romain, 2022-03-30:
`lidR::plot(points_georeferenced@data)` is not possible it actually calls `plot.data.frame()` by inheritance which indeed runs forever (at least irrelevantly long) for a large point cloud. It plots every attribute against every attributes try

```r
data(iris)
plot(iris)
```

And imagine what it can do with a lidar point cloud. The way to work is `plot(points_georeferenced)` and you never need to call `@data`.

cingularities, 2022-03-30:
the fatal error occurs on lidR::plot(points_georeferenced)

cingularities, 2022-03-30:
#459 I am also having a similar issue with this case. The exact same error when I run normalize_height. I already tried: filter = "-drop_class 7" with no success

cingularities, 2022-03-30:
[](https://drive.google.com/file/d/17nSTTwp8Mv98-O7rFBTc92Z896aJQtpS/view?usp=sharing)

cingularities, 2022-03-30:
[https://drive.google.com/file/d/17nSTTwp8Mv98-O7rFBTc92Z896aJQtpS/view?usp=sharing](url)

Jean-Romain, 2022-03-30:
> https://github.com/r-lidar/lidR/discussions/459 I am also having a similar issue with this case

If you have an issue with normalize_height please open a dedicated issue with a reproducible example.

> the fatal error occurs on `lidR::plot(points_georeferenced)` and aborts my session

I see that you are running `lidR 3.2.4`. You are not up to date with lidR so I guess you are not up-to-date with other packages as well. Before to report a bug please ensure that all your packages are up-to-date. The problem may disappear by itself. But I'm very unsure it will in this case.

Jean-Romain, 2022-04-20:
Can you plot a point cloud from small datasets included with the package?

AndresKasekamp, 2022-04-27:
I also have problem with reading in laz files from one of the flight years from my NMA. It causes RStudio to crash without any messages. All the other years work perfectly and I can easily apply functions to it. Using the erroneous laz files in LASTools has not produced any errors. I am using the latest lidR version.

I am attaching a Google Drive link with the error and no error laz files.
https://drive.google.com/drive/folders/1SVsfenH_petxogLg7-QvBaeQ98aHrDvj?usp=sharing

Jean-Romain, 2022-04-27:
@AndresKasekamp  please open an issue. This is by no mean related to the 3D rendering issue reported here.

closed 2022-04-29 by Jean-Romain