r-lidar/lidR #558: rasterize_terrain() with sfc shape

wiesehahn, 2022-03-30:
In the documentation of `rasterize_terrain()` it says for `shape` argument: "It can also be an sfc to define a polygon in which to perform the interpolation."

In my understanding this means that I can pass a catalog and some region as sfc to get a dtm for that region. However, this seems not to work. E.g.:
```
# A single file LAScatalog using data provided with the package
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)

bbox <- st_bbox(ctg@data[1,])

dtm <- rasterize_terrain(ctg, res=1, shape= st_as_sfc(bbox))
```
outputs 
```
Processing [==============================================================] 100% (1/1) eta:  0s
unable to find an inherited method for function ‘mosaic’ for signature ‘"SpatRaster", "missing"’Warning message:
An error occured during the automatic merge of 'catalog_apply'. Merging is impossible. A list has been returned. 
```

Is this an error or am I understanding it wrong?

Jean-Romain, 2022-03-30:
> Is this an error or am I understanding it wrong?

I think both ! 

First this is an error, it should work and actually it is related to the fact you are processing a single file. It is reproductible without `shape`

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
dtm = rasterize_terrain(ctg, 1, tin())
```

Second, your example seems useless to me. Please exaplain what you have in mind when using `shape= st_as_sfc(bbox)` so I can see if something can be inproved in the documentation. Thank.

Jean-Romain, 2022-03-30:
Fixed

closed 2022-03-30 by Jean-Romain

wiesehahn, 2022-03-30:
Ok, so my initial intention was to skip the step to clip the catalog  (`las <- clip_roi(ctg, bbox)`), but instead definde the area of the requested dtm inside `rasterize_terrain()`

e.g.
```
ctg <- readLAScatalog(folder)

# get area as bbox from one of the las files
bbox <- st_bbox(ctg@data[1,])

dtm <- rasterize_terrain(ctg, res=1, shape= st_as_sfc(bbox))
```

this results in
```
Processing [---------------------------------------------------------------------------------------------------------]   0% (1/49588) eta:  8dError: subscript out of bounds
```

Jean-Romain, 2022-03-30:
> Ok, so my initial intention was to skip the step to clip the catalog (`las <- clip_roi(ctg, bbox)`)

That makes sense and looks like a perfectly valid use of the argument `shape` if the `shape` is smaller than the tiles or if the shape is complex. But if the `shape` is the bounding box of the single tile it is equivalent as not providing anything.

> Error: subscript out of bounds

Fixed. Especially it did not behave as expected. Thank you for reporting

commit 8e73a416 (2022-03-30): Fix #558
commit c30ea50c (2022-03-30): Fix #558