r-lidar/lidR #559: metrics_template.R uses terra::set.value(), which does not exist

corneliussenf, 2022-03-31:
Dear all,

I am running pixel_metrics() on a las collection and I get an error that terra::set.value() does not exist. I did some digging and found that the function is used in the metrics_template.R file. After checking with the terra package, it seems that indeed set.values() is not available in terra, but setValues() is. Is this is bug or am I missing something here?

Best

Cornelius

Jean-Romain, 2022-03-31:
Please report a minimal reproducible example. I'm using `pixel_metrics` on daily basis without trouble.

Jean-Romain, 2022-03-31:
An I confirm that `terra` do have `set.values`. I'm using 1.5-21

corneliussenf, 2022-03-31:
Sure, sorry, I thought the issue would be identifiable without a short example, but you're correct, I should have added an example in the first place. The following code-snipped using the LAS file attached (zipped) does produce the following error: Error: 'set.values' is not an exported object from 'namespace:terra'.

`nlas <- lidR::readLAS("671750_5262250.las")`
`metrics <- pixel_metrics(nlas, func = .stdmetrics)`

[671750_5262250.las.zip](https://github.com/r-lidar/lidR/files/8389134/671750_5262250.las.zip)

Session information:

R version 3.6.3 (2020-02-29)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 18.04.5 LTS

Matrix products: default
BLAS:   /usr/lib/x86_64-linux-gnu/blas/libblas.so.3.7.1
LAPACK: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.7.1

locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C               LC_TIME=en_US.UTF-8        LC_COLLATE=en_US.UTF-8     LC_MONETARY=en_US.UTF-8   
 [6] LC_MESSAGES=en_US.UTF-8    LC_PAPER=en_US.UTF-8       LC_NAME=C                  LC_ADDRESS=C               LC_TELEPHONE=C            
[11] LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C       

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
 [1] terra_1.4-22    future_1.21.0   sf_1.0-2        forcats_0.5.1   stringr_1.4.0   dplyr_1.0.4     purrr_0.3.4     readr_1.4.0    
 [9] tidyr_1.1.2     tibble_3.1.6    ggplot2_3.3.3   tidyverse_1.3.0 lidR_4.0.0     

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.6         lubridate_1.7.9.2  lattice_0.20-41    listenv_0.8.0      prettyunits_1.1.1  class_7.3-17       assertthat_0.2.1  
 [8] digest_0.6.27      utf8_1.2.2         parallelly_1.23.0  R6_2.5.0           cellranger_1.1.0   backports_1.2.1    reprex_1.0.0      
[15] e1071_1.7-4        httr_1.4.2         pillar_1.6.4       progress_1.2.2     rlang_0.4.12       lazyeval_0.2.2     readxl_1.3.1      
[22] rstudioapi_0.13    data.table_1.13.6  raster_3.4-13      rgdal_1.5-23       munsell_0.5.0      broom_0.7.4        compiler_3.6.3    
[29] modelr_0.1.8       pkgconfig_2.0.3    globals_0.14.0     tidyselect_1.1.0   codetools_0.2-16   fansi_0.5.0        crayon_1.4.2      
[36] dbplyr_2.1.0       withr_2.4.1        grid_3.6.3         jsonlite_1.7.2     lwgeom_0.2-5       gtable_0.3.0       lifecycle_1.0.1   
[43] DBI_1.1.1          magrittr_2.0.1     units_0.6-7        scales_1.1.1       KernSmooth_2.23-17 cli_3.1.0          stringi_1.5.3     
[50] fs_1.5.0           sp_1.4-5           xml2_1.3.2         ellipsis_0.3.2     generics_0.1.0     vctrs_0.3.8        tools_3.6.3       
[57] glue_1.5.1         hms_1.0.0          abind_1.4-5        parallel_3.6.3     yaml_2.2.1         colorspace_2.0-0   rlas_1.5.0        
[64] stars_0.5-1        classInt_0.4-3     rvest_0.3.6        haven_2.3.1

Jean-Romain, 2022-03-31:
It was added in 1.5-17. I change the `DESCRIPTION` file to depends on version >= 1.5-17. Thanks

closed 2022-03-31 by Jean-Romain

corneliussenf, 2022-03-31:
Perfect, thanks. It now works!