r-lidar/lidR #563: error in retile_catalog when using flight lines

floresteiro, 2022-04-06:
Hi,

I am trying to retile flight transects using catalog_retile, but the following error arises: 

> > newctg2 = catalog_retile(ctg2)
>   |====================================================================| 100%
> missing value where TRUE/FALSE neededWarning message:
> An error occured during the automatic merge of 'catalog_apply'. Merging is impossible. A list has been returned.

It worked when I used only one flight transect tho. It also worked when I used a dataset of tiles.

Here is the code I used (which is basically the original example from the retile a LAScatalog documentation):

```
library(lidR)

# catalog with 2 flight lines

ctg = readLAScatalog("Test_2flight_lines")

plot(ctg)

# Create a new set of 500 x 500 m .las files

opt_chunk_buffer(ctg) <- 0
opt_chunk_size(ctg) <- 500                       
opt_filter(ctg) <- ""                              
opt_laz_compression(ctg) <- TRUE
opt_chunk_alignment(ctg) <- c(0,0) 
opt_output_files(ctg) <- paste0(tempdir(), "/retile_{XLEFT}_{YBOTTOM}") 

# preview the chunk pattern
plot(ctg, chunk = TRUE)

newctg = catalog_retile(ctg)

plot(newctg)
```
Thank you!

Jean-Romain, 2022-04-06:
I tested the following and it works. It means there is something else, maybe something specific with your data. Please share a minimal reproducible example including data if necessary

```r
library(lidR)

# catalog with 2 flight lines
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
plot(ctg)

# Create a new set of 500 x 500 m .las files
opt_chunk_buffer(ctg) <- 0
opt_chunk_size(ctg) <- 300                       
opt_filter(ctg) <- ""                              
opt_laz_compression(ctg) <- TRUE
opt_chunk_alignment(ctg) <- c(0,0) 
opt_output_files(ctg) <- paste0(tempdir(), "/retile_{XLEFT}_{YBOTTOM}") 

# preview the chunk pattern
plot(ctg, chunk = TRUE)

newctg = catalog_retile(ctg)
plot(newctg)
```

Jean-Romain, 2022-04-06:
And please report what `newctg2` contains. It should be a list of strings.

floresteiro, 2022-04-08:
It worked here, I received that warning message when testing with two flight lines in which the overlap area had some small holes that I hadn't noticed. 

`newctg2 ` contains a list
[image]

So when I tested again using two flight lines that overlap perfectly, I didn't receive that message and the output (`newctg2b`) contained a large LAScatalog:
[image]

Then I ran the whole dataset. As a result, I still received that same warning message and, similar to `newctg2`, the output contained a list. Despite this warning message, the input was retiled as I wanted, so it does not seem to be a problem. 

Thank you very much for taking your time to check my issue, I really appreciate it!

Jean-Romain, 2022-04-08:
There is a problem if it does not work as expected. And it doesn't. Could you please share a minimal reproducible example

floresteiro, 2022-04-08:
Ok, here are the flight lines from my problematic test:
 
https://drive.google.com/drive/folders/1QD_6kpDW9LpF1iyNcOq2qqfV6lwSSG4s?usp=sharing

Jean-Romain, 2022-04-08:
The link has restricted access

floresteiro, 2022-04-08:
Sorry, you should be able to access it now

Jean-Romain, 2022-04-08:
I fixed the bug. It came from the fact that some chunks were empty. Thanks for reporting.

[image]

By the way your dataset was not spatially indexed so the retiling process took me 15 minutes while with a spatial index it took me 30 seconds. You should always consider indexing your tiles

commit d9b24ad0 (2022-04-08): Fix #563
closed 2022-04-08 by Jean-Romain