r-lidar/lidR #564: sub-meter resolution for grid_metrics or pixel_metrics

stukrause, 2022-04-07:
I am working with really high resolution photogrammetric las and would like to create sub-meter resolution rasters with the pixel_metrics function. Any res values under 3 throw an error. Is there a way around this? I am assuming it is set up only for integers...

stukrause, 2022-04-07:
Ah Sorry, grid_metrics works just fine with lower resolutions, also sub-meter. It seems to be the lidRmetrics package. Using a resolution lower than 15 throws an error with the example dataset. I will bring it up with them. 

I am actually just trying extract OpenGapSpace and CloseGapSpace so I can calculate gap fraction on a grid at around 0.3 meters resolution. lidR just has the area-based gap_fraction_profile right?

`devtools::install_github("ptompalski/lidRmetrics")
library(lidR)
library(lidRmetrics)

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile)

m3 <- grid_metrics(las, ~metrics_set2(X, Y, Z), res = 10)`

Jean-Romain, 2022-04-07:
I get 

```r
 m3 <- grid_metrics(las, ~metrics_set2(X, Y, Z), res = 10)
#> Erreur dans `[.data.table`(las@data, , if (!anyNA(.BY)) metrics_set2(X, Y,  : 
#>   Column 68 of result for group 540 is type 'double' but expecting type 'integer'. Column types must be consistent for each group.
#> De plus : Il y a eu 50 avis ou plus (utilisez warnings() pour voir les 50 premiers)
```

Which is an issue with `metrics_set2` which is not from `lidR`

closed 2022-04-07 by Jean-Romain