r-lidar/lidR #566: catalog_retile does not maintain original filenames

azh2, 2022-04-10:
**I am retiling my .las catalog in order to apply a custom algorithm and use the file names as an index to call in matching layers. I applied the following code to the catalog:**

library("lidR")
dir.create(paste0(tmpdir, "\\RETILE_LAS"))
LidRtst <- lidR::readLAScatalog(folder = paste0(tmpdir, "\\NDVI_FILT"))
lidR::opt_output_files(LidRtst) <- paste0(tmpdir, "\\RETILE_LAS\\{ORIGINALFILENAME}_RETILE")
opt_laz_compression(LidRtst) <- TRUE
future::plan(strategy = future::multisession, workers = parallel::detectCores()-1)
lidR::set_lidr_threads(parallel::detectCores()-1)
lidR::opt_wall_to_wall(LidRtst) <- TRUE 
lidR::opt_chunk_buffer(LidRtst) <- 100
lidR::opt_progress(LidRtst) <- TRUE
lidR::opt_merge(LidRtst) <- FALSE
lidR::opt_stop_early(LidRtst) <- TRUE
lidR::catalog_retile(LidRtst)
gc()

**And then created a catalog overlay index using a custom function and imported this into QGIS**

[image]

In the image above, the original catalog index is in green and the retiled index is in blue, the names of the selected bounding boxes should match, but they don't, the original has a file ID of 006307 and the retiled file has an ID of 006298. The first thousand or so filenames matched, but by the time we get to the end of the catalog which includes 6,200 files, none of them matched.

Jean-Romain, 2022-04-11:
Well, could you try to help me to narrow down the problem. I mean I can't run `catalog_retile` during hours to **maybe** reproduce. Run the following and tell me if some input filenames are different from the output

```r
folder = paste0(tmpdir, "\\NDVI_FILT")

library(lidR)

LidRtst <- lidR::readLAScatalog(folder)
lidR::opt_output_files(LidRtst) <- "{ORIGINALFILENAME}"
opt_laz_compression(LidRtst) <- TRUE
lidR::opt_wall_to_wall(LidRtst) <- FALSE 
lidR::opt_chunk_buffer(LidRtst) <- 100
lidR::opt_progress(LidRtst) <- TRUE
lidR::opt_merge(LidRtst) <- FALSE
lidR::opt_stop_early(LidRtst) <- TRUE

chunks = engine_chunks(LidRtst)
ans = lapply(chunks, function(x) return(list(i = basename(tools::file_path_sans_ext(x@files[1])), o = basename(x@save))))
ans = data.table::rbindlist(ans)
any(ans$i != ans$o)
```

azh2, 2022-04-12:
I get the following output:

any(ans$i != ans$o)

[1] TRUE

There are 5035 records where i and o do not match. So this would indicate the mismatch occurred prior to retiling the catalog?

Jean-Romain, 2022-04-12:
> So this would indicate the mismatch occurred prior to retiling the catalog?

It appends during the preprocessing step when creating the chunks. The good new is that you can save `LidRtst` with `saveRDS` and share it to me. I'll be able to reproduce without the physical point cloud.