r-lidar/lidR #568: raster_as_las bbox method missing

Silviculturalist, 2022-04-11:
Hi,

When loading a .tif raster (on-disk) through terra and running `locate_trees`, it seems like there is no function which evaluates the bbox before handing it over to raster_as_las(). This causes it to immediately fail, as default bbox=NULL. 

I think there is a method missing to find or state the bbox for the raster.

Best,
Carl

Example code (TIF not uploadable to GitHub, can send via email).
```
#load example raster
input_raster <- terra::rast(x = "file_name_here.tif")
lidR::locate_trees(input_raster,algorithm = "lmf",uniqueness = "incremental")
>Error: On-disk rasters not supported with a bbox in 'raster_as_las()'.
```

Silviculturalist, 2022-04-11:
If I force the raster into memory following your update e.g. b95c0ae31e705612b93e8d86cf489843bbb7031c LOC: 636 ```(raster*1)```

I instead get issue that neither 'lmf' or 'manual' are provided as algorithm. They seem to be missing from lidR:::LIDRALGORITHM ?  e.g. `lidR::is.algorithm()`.

Suggestion that 'lmf' is set to default?

Jean-Romain, 2022-04-11:
In your first message, the error is expected. Yet it is not triggered as early as it should and not with the most explicit message. I can improve that. I don't understand the second message. Please report a minimal reproducible example.

Silviculturalist, 2022-04-11:
Maybe this is clearer? It's the 'next error along the line'.

```
input_raster <- terra::rast(x = "file_name_here.tif")
lidR::locate_trees(input_raster,algorithm = "lmf",uniqueness = "incremental")
>Error: On-disk rasters not supported without a bbox in 'raster_as_las()'

input_raster <- input_raster*1
locate_trees(input_raster,algorithm = "lmf",uniqueness = "incremental")
>Error: Invalid function provided as algorithm.
```

Silviculturalist, 2022-04-11:
Works perfectly now, thank you. My colleagues were also wondering what was going wrong, and kept stating that the new lidR was working so much better than the old version! Many thanks, keep up the great work.

Silviculturalist, 2022-04-11:
I can try the dev. version and see how that works.

closed 2022-04-11 by Silviculturalist