r-lidar/lidR #571: Error in fast_quantization(las@data[["Z"]], zscale, zoffset)

cingularities, 2022-04-19:
Running normalized height.

```AGL <- normalize_height(canopy_points, DTM, na.rm = TRUE, copy=TRUE)```

Error:
```
Error in fast_quantization(las@data[["Z"]], zscale, zoffset) : 
  Non quantizable value outside the range of representable values of type 'int'
```

I already loaded the file using filter = "-drop_class 7"

lidR version: 4.0.0

cingularities, 2022-04-19:
[https://drive.google.com/file/d/1Y8apzOpm3PoEM-Tun9r9vI4s1AzotvKi/view?usp=sharing](url)

Jean-Romain, 2022-04-20:
Can you investigate to reduce the problem to a smaller file/area. Downloading 3 GB will take me hours and I can't load such big file on my computer. And you did non share the DTM. If you computed the DTM with `rasterize_terrain()` please share a minimal reproducible example.

cingularities, 2022-04-20:
[https://drive.google.com/drive/folders/12Z-vexz_cRNlgkHZLhkv9YIPxNu-_xQ7?usp=sharing](url)
Here you go.
Folder includes canopy points and DTM

Jean-Romain, 2022-04-20:
Reproduced with 

```r
library(lidR)
library(terra)
setwd("~/Téléchargements/lidR issues/issue 571/")

dtm = rast("DTM_042022.tif")
las = readLAS("canopy_points_042022.las")
nlas <- normalize_height(las, dtm*1)
```

Notice that `na.rm` is no longer useful since v4.0.0 and `copy`,as far as I can remember, never existed. Anyway it does not change anything to the issue.

Jean-Romain, 2022-04-20:
The Z scale factor is not set to 0 when normalizing. Thanks for reporting.

commit 6f5cdd42 (2022-04-20): Fix #571 Z offset set to 0
closed 2022-04-20 by Jean-Romain