r-lidar/lidR #574: R crashes when reading in laz files with lidR

AndresKasekamp, 2022-04-27:
I have problem with reading in laz files from one of the flight years from my NMA. It causes RStudio to crash without any messages. All the other years work perfectly and I can easily apply functions to it. Using the erroneous laz files in LASTools has not produced any errors. I am using the latest lidR version.

It is probably related to https://stackoverflow.com/questions/60457539/lidr-readlas-function-crashes-r-when-reading-some-las-files . 

I am attaching a Google Drive link with the error and no error laz files.
https://drive.google.com/drive/folders/1SVsfenH_petxogLg7-QvBaeQ98aHrDvj?usp=sharing

Jean-Romain, 2022-04-27:
Sorry, I inadvertently deleted the file instead of downloaded it. I can't go back. Can you upload again.

AndresKasekamp, 2022-04-27:
Uploaded again.

Jean-Romain, 2022-04-27:
This is not related to `lidR`. I'm moving to `rlas`

```r
library(rlas)
las = read.las("issue 574/no_error.laz")
```