r-lidar/lidR #576: Issue with 'merge_spatial'

Francoisdt14, 2022-05-02:
Hi JR,

I have been trying to use 'merge_spatial' to merge known tree crown boundaries for a tree improvement trial (with tree information and a tree ID) with a las file. I was able to do this for my study area in the past (approximately 1500 trees using lidR version 3), but have not been able to consistently do it since updating to the latest version (4.0.0). I tried selecting different subsets of tree crown polygons and found that some subsets worked, while some failed. When the function fails, the message says: 'Error: Only spatial polygons are supported', however, I am not sure why the error is occurring. 

See the REPREX below. I have included 2 shape files ('success' and 'fail') with example trees, along with a las tile in the zipped folder below.

```
  library(lidR)
  library(sf)
  
  success <- st_read("D:/lidR_Issue/success.shp")
  fail <- st_read("D:/lidR_Issue/fail.shp")
  las <- readLAS("D:/lidR_Issue/test_tile.laz")
  
  las_test = merge_spatial(las, success, "treeID") # works
  las_test2 = merge_spatial(las, fail, "treeID") #fails

```

[lidR_Issue.zip](https://github.com/r-lidar/lidR/files/8605842/lidR_Issue.zip)

Jean-Romain, 2022-05-02:
I can't test right know but let me try to make guess:  `succes` contains `POLYGONS` and `fails` contains `POLYGONS` and `MULTIPOLYGONS` and its type is `GEOMETRY`. Am I correct?

Francoisdt14, 2022-05-02:
Hi JR, 

It seems that is not correct - both are multipolygons.

`> unique(st_geometry_type(success))
[1] MULTIPOLYGON
18 Levels: GEOMETRY POINT LINESTRING POLYGON MULTIPOINT MULTILINESTRING MULTIPOLYGON GEOMETRYCOLLECTION CIRCULARSTRING COMPOUNDCURVE CURVEPOLYGON MULTICURVE MULTISURFACE CURVE ... TRIANGLE
> unique(st_geometry_type(fail))
[1] MULTIPOLYGON
18 Levels: GEOMETRY POINT LINESTRING POLYGON MULTIPOINT MULTILINESTRING MULTIPOLYGON GEOMETRYCOLLECTION CIRCULARSTRING COMPOUNDCURVE CURVEPOLYGON MULTICURVE MULTISURFACE CURVE ... TRIANGLE`

After using `st_cast` on both the `success` and `fail` polygons `(test <- st_cast(fail, "POLYGON")`) the error persists for fail.

Jean-Romain, 2022-05-02:
I think it is related to the fact the polygons are MULTIPOLYGONS instead of POLYGONS and I think I already fixed it in a previous commit because I can't reproduce.

Francoisdt14, 2022-05-03:
Hi JR - thanks for the quick replies. I installed the latest version from GitHub, and it worked! I was on the latest CRAN version so thanks for the tip.

Jean-Romain, 2022-05-03:
I going to release v4.0.1 today or tommorow

closed 2022-05-05 by Jean-Romain