r-lidar/lidR #577: Problem with plotting CHM error with "is.bool" function not found

ecoinfo-netra, 2022-05-03:
I tried an example with CHM and DSM . I want to plot it but it is giving me an error with "is.bool" function.
Is there any another library that should be installed?
Example given here - 

LASfile <- system.file("extdata", "Topography.laz", package = "lidR")
las2 <- readLAS(LASfile)
las2 <- normalize_height(las2, algorithm = tin())
plot(las2, size = 3, bg = "white")
chm <- rasterize_canopy(las2, res = 0.5, algorithm = dsmtin())
plot(chm, col = col)
Error in is.bool(x) : could not find function "is.bool"

Jean-Romain, 2022-05-03:
I cannot reproduce. I suggest reinstalling lidR. Maybe something went wrong during the installation

ecoinfo-netra, 2022-05-03:
Thanks I worked it out now . I need to install terra and gstat to make it work with lidR