r-lidar/lidR #580: normalize_height causes "NULL value passed as symbol address"

Jean-Romain, 2022-05-06:
### Discussed in https://github.com/r-lidar/lidR/discussions/579

<div type='discussions-op-text'>

<sup>Originally posted by **kulpojke** May  5, 2022</sup>
I am using the lidR package to do some tree segmentation on a set of 561 km² tiles using the `LAScatalog` engine and the futures library for parallelization. I have successfully created the DTM, but when I attempt to normalize I get an error.

I have read [this thread](https://github.com/r-lidar/lidR/discussions/547) already and it did not solve my problem.

I am running inside of a container (`rocker/tidyverse:4.2`) with the following:

```
R version 4.2.0 (2022-04-22)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 20.04.4 LTS

future_1.25.0
lidR_4.0.0
```

Here is the code:

```r
# import libraries
library(lidR)
library(future)

plan(multisession)

# define some paramaters, currently using ft :(
chunk_size <-1000            # chunk size 
buff <- 50                   # buffer size
resolution <-2               # raster resolution

# (1) path to data 
path <- '/data'

# make las catalog
ctg <- readLAScatalog(path)

# set disk storage option by assigning a filename template to a function
opt_output_files(ctg) <- paste0(tempdir(), "/{XCENTER}_{YCENTER}_{ID}")

# create index files (lax) to index points within laz files. speeds up reads.
lidR:::catalog_laxindex(ctg)

# set the chunk buffer ofr ctg
opt_chunk_buffer(ctg) <- buff

# set the chunk size for ctg
opt_chunk_size(ctg) <- chunk_size

# make dtm
opt_output_files(ctg) <- paste0(tempdir(), "/{XCENTER}_{YCENTER}_{ID}_dtm")
dtm <- rasterize_terrain(ctg, resolution, tin())

# in my Rmd I plot the dtm here and it looks right

# normalise ctg
opt_output_files(ctg) <- paste0(tempdir(), "/{XCENTER}_{YCENTER}_{ID}_norm")
ctg_norm <- normalize_height(ctg, dtm)

```

During the normalization I get the following warning, then error:

```
Processing [-----------------]   0% (0/1152) eta:  ?sWarning: There are 3 points flagged 'withheld'.
Processing [-----------------]   0% (1/1152) eta:  4hError: NULL value passed as symbol address
```

After reading [this thread](https://github.com/r-lidar/lidR/discussions/547). I tried disabling multisession and using set_lidr_threads,

```
# import libraries
library(lidR)
library(future)

#plan(multisession)
get_lidr_threads()
```
but the problem persists. Any insights would be greatly appreciated.
</div>

Jean-Romain, 2022-05-06:
After reproducing with data of mine I get 

```r
Error : external pointer is not valid
```

Which is likely to be very similar to your error. `SpatRaster` are not serializable and I think this is what caused the error. However I do not have error with sequential processing because only one worker is implied and no serialization is involved. Are you sure you really disabled `plan(multissesion)`. Did you start a fresh session or reset the strategy with `plan(sequential)`?

Side note: `plan(multisession)` and  `s/get_lidr_threads()` are doing very different things.

kulpojke, 2022-05-08:
You were correct. I forgot to restart the session.  I restarted it and it worked without the multi-threading.  Does this mean parallelization is not possible on this task currently?

Jean-Romain, 2022-05-08:
You have several workaround
1. convert to `RasterLayer`
2. convert to `stars`
3. write the DTM on disk and use on-disk `SpatRaster` (not sure this one works)

kulpojke, 2022-05-10:
It creates the dtm using stars but when I try to normalize, 

```
Error: On-disk rasters not supported in 'raster_as_dataframe()'
```

Apparently  `st_as_stars()` makes  a  `stars_proxy` (which makes sense for converting an on disk raster), but it looks like lidR does not work with the `stars_proxy` object based on

```
#' @importFrom stats na.omit
[raster_as_dataframe](https://rdrr.io/cran/lidR/src/R/utils_raster.R) <- [function](https://rdrr.io/r/base/function.html)(raster,  xy = [TRUE](https://rdrr.io/r/base/logical.html), na.rm = [TRUE](https://rdrr.io/r/base/logical.html))
{
  if ([raster_is_proxy](https://rdrr.io/cran/lidR/src/R/utils_raster.R)(raster))
    [stop](https://rdrr.io/r/base/stop.html)("On-disk rasters not supported in 'raster_as_dataframe()'", call. = [FALSE](https://rdrr.io/r/base/logical.html)) # nocov
```
So I guess I will try `RasterLayer`

Jean-Romain, 2022-05-10:
It is hard to test every single case 3 times, one per package. It should work but working with stars has been less extensively tested. The best is to report troubleshooting.

Jean-Romain, 2022-05-24:
Fixed with a workaround. If user is using a parallel strategy `SpatRaster` are converted to `RasterLayer`. There is not way to work with `SpatRaster`

closed 2022-05-24 by Jean-Romain

Ben-Shamgochian, 2022-09-15:
> Fixed with a workaround. If user is using a parallel strategy `SpatRaster` are converted to `RasterLayer`. There is not way to work with `SpatRaster`

Hi, I'm currently experiencing the same issue when attempting to normalize height and I don't quite understand this work around. Do you mind explaining in a little more depth?

Thanks!

commit cd9e6c9a (2022-05-24): Fix #580 by converting to serializable RasterLayer