r-lidar/lidR #583: Clarification on the dsmtin algorithm

floresteiro, 2022-05-16:
Hi Jean-Romain,

I tried to use` dsmtin` to generate a DSM similar to the one I can get when I use only first returns in lastools' las2dem (

> this tool reads LIDAR points from the LAS/LAZ format, triangulates lidar points temporarily into a TIN, and then rasters the TIN onto a DEM

), because it seemed to work like (similarly) las2dem works. However, the output of `dsmtin`  is not as detailed as that of las2dem's  and looks more like the outputs of algorithms that use the highest point found in each pixel, such as `p2r` or the lasgrid tool (with the -highest setting) on lastools. 

I am particularly interested in a DSM that does not remove/ignore lower first returns and, therefore, provides a more detailed crown (rougher canopy). In this regard, If possible, I would like to know if I can get a DSM using `dsmtin `that reflects the first return penetration in the canopy, such as the one I got with las2dem.

 In the figure below, the left side shows the output of `dsmtin` and the right side shows the output of las2dem.
[image]
The mapswipe tool, on QGIS, was used to divided the same area in two. Therefore, the picture shows adjacent areas.

Jean-Romain, 2022-05-16:
According to the doc of `dsmtin`

> This function is made to be used in rasterize_canopy. It implements an algorithm for digital surface model computation using a Delaunay triangulation of **first returns** with a linear interpolation within each triangle.

There is no mention of *first returns* in `las2dem`. Did you called `las2dem` with a first return filter?

floresteiro, 2022-05-17:
Yes, I called it with a first return filter. So, I was expecting that they would produce similar outputs.

Jean-Romain, 2022-05-17:
Can you run the same on the example file included with lidR <kbd>Megaplot.laz</kbd>. Show me the `lidR` output + the R source code + the `lastools` output + the lastools command. Remember that `lastools` is close source. We can't know exactly what it does. Maybe the rasterization step is different. But I'll try to provide an explanation and an option in lidR to reproduce more closely `lastools` output.

floresteiro, 2022-05-17:
`lidR` R source code:
```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)
dsm = rasterize_canopy(las, res=0.5, algorithm = dsmtin())
writeRaster(dsm,"D:/Megaplot_50cm_dsmtin.tif")
```
`lastools `command:
```
las2dem -i D:\Megaplot.laz -o d:\Megaplot_dsm_50cm_las2dem.tif -otif^ 
		-first_only ^
		-step 0.5 ^
		-elevation 
```

In the picture, the left side shows the output of `dsmtin` and the right side shows the output of` las2dem`.
[image]
The mapswipe tool, on QGIS, was used to divided the same area in two. Therefore, the picture shows adjacent areas.

Jean-Romain, 2022-05-17:
I'm not able to answer. One step of `dsmtin` consists in keeping only the highest point per pixel before the triangulation to reduce computation time. This is not documented and I will add and option to include all points and I will update the doc. Yet if I produce a CHM skipping this step I still do not have a `las2dem`-like CHM.

```r
dsm1 = rasterize_canopy(las, res=0.5, algorithm = dsmtin(highest = TRUE))
dsm2 = rasterize_canopy(las, res=0.5, algorithm = dsmtin(highest = FALSE))
```

[image]

I can hardly see were those `las2dem` pits are coming from. Look at the 1-m resolution point-to-raster CHM. There is no extra pits

```r
dsm3 = rasterize_canopy(las, res=1, algorithm = p2r())
```
[image]