r-lidar/lidR #584: Error al realizar lidR::normalize_height

carlostp96, 2022-05-21:
Buenos días, soy nuevo en esta página y no se si estoy realizando en el sitio adecuado la consulta. La expongo de todos modos.

Estoy realizando un código para mi trabajo de fin de master, en una parte del código busco obtener la normalización de los DSM de algunos ayuntamientos de Galicia. Los archivos LiDAR empleados proceden del CNIG, son los datos de la 2ª cobertura (2015-actualidad), los datos de los DTM proceden también del CNIG y son los datos con un paso de maya de 2x2. La información LiDAR por ayuntamientos es muy pesada y he tenido que llevar a cabo la lectura de los archivos .las por partes.

Después de la lectura de los archivos calculo el DSM con la siguiente función:
DSM_estrada_1 <- lidR::grid_canopy(LiDARfiles_estrada_1, res = 2, algorithm = lidR::p2r(subcircle = 1, na.fill = tin()))

Este primer paso funciona perfectamente en cualquiera de los datos seleccionados, el problema surge al normalizar.

Al realizar la función para alguno de los ayuntamientos seleccionados me sale el siguiente error:
> LiDARNorm <- lidR::normalize_height(LiDARfiles_estrada_1, InpDTM, na.rm = TRUE)
Error: On-disk rasters not supported in 'raster_as_dataframe()'
In addition: Warning message:
In raster_cell_from_xy(raster, x, y) :
  NAs introduced by coercion to integer range

EL tema es que en algunos ayuntamientos si me funciona y en otros me sale ese error. No consigo encontrar por ningún lado una posible solución, no se si alguien podría ayudarme?

Un saludo y muchas gracias de antemano

Jean-Romain, 2022-05-21:
I do not speak spanish

carlostp96, 2022-05-22:
Good morning, I'm new to this page and I don't know if I'm making the query in the right place. Exposure in all ways.

I am making a code for my master's thesis, in a part of the code I seek to obtain the normalization of the DSM of some municipalities of Galicia. The LiDAR files used come from the CNIG, they are the data from the 2nd coverage (2015-current), the DTM data also come from the CNIG and they are the data with a mesh step of 2x2. The LiDAR information by municipalities is very heavy and I have had to carry out the reading of the .las files in parts.

After reading the files I calculate the DSM with the following function:
DSM_estrada_1 <- lidR::grid_canopy(LiDARfiles_estrada_1, res = 2, algorithm = lidR::p2r(subcircle = 1, na.fill = tin()))

This first step works perfectly on any of the selected data, the problem arises when normalizing.

When performing the function for any of the selected municipalities I get the following error:

LiDARNorm <- lidR::normalize_height(LiDARfiles_estrada_1, InpDTM, na.rm = TRUE)
Error: On-disk rasters not supported in 'raster_as_dataframe()'
In addition: Warning message:
In raster_cell_from_xy(raster, x, y) :
NAs introduced by coercion to integer range

The issue is that in some municipalities it works for me and in others I get that error. I can't find a possible solution anywhere, I don't know if someone could help me?

Thank you very much in advance

closed 2022-05-30 by Jean-Romain

Silviculturalist, 2024-03-11:
@Jean-Romain @carlostp96 I am fairly certain from my own testing in my script that this error, which I've had also, is created by terra being loaded after lidR. This causes some functions to be masked. Simply changing the order in which the packages were loaded resolved the error: 'On-disk rasters not supported in 'raster_as_dataframe()'. 

Although I have not written up a reproducible example, I'm fairly certain you would get this result after e.g.:
i) load lidR, then load terra.
ii) load a lascatalogue.
iii) creating a relative to the chunks high-resolution DTM and saving this to several geotiffs.
iv) create a VRT of the DTM with terra.
v) import the VRT with raster::raster
vi) pass the vrt and the lascatalogue to normalize_height.

Jean-Romain, 2024-03-11:
The error comes from the fact that, by default, terra does not load the data in memory. Data must be loaded in memory first